ros-drivers/rosserial #341: Cann't receive /map data

wbang8933, 2017-12-06:
I can receive odom data but cantn't get /map data in windows.

`#include "stdafx.h"
#include <string>  
#include <stdio.h>  
#include "ros.h"
#include <nav_msgs/OccupancyGrid.h>
#include <nav_msgs/Odometry.h>
#include <nav_msgs/MapMetaData.h>
#include <windows.h> 

using std::string;

void map_callback(const nav_msgs::OccupancyGrid &map)
{
	printf("Received map:\n");
	printf("\tmap info: %d * %d, r(%f)\n", map.info.width,
		map.info.height, map.info.resolution);
	
}

void metaData_callback(const nav_msgs::MapMetaData &meta)
{
	printf("Received MapMetaData...\n");
}

void odom_callback(const nav_msgs::Odometry &odom)
{
	printf("Received odometry...\n");
	printf("odom data:\n");
	printf("\tpose:[%f,%f,%f]\n", odom.pose.pose.position.x, odom.pose.pose.position.y, odom.pose.pose.position.z);
	printf("\torient:[%f,%f,%f,%f]\n", odom.pose.pose.orientation.x, odom.pose.pose.orientation.y, odom.pose.pose.orientation.z,
		odom.pose.pose.orientation.w);
}

int _tmain(int argc, _TCHAR* argv[])
{
	ros::NodeHandle nh; 
	char *ros_master = "10.10.14.40";  

	printf("Connecting to server at %s\n", ros_master);
	nh.initNode(ros_master);

	printf ("Advertising map message\n");

	ros::Subscriber < nav_msgs::OccupancyGrid > mapSub("map", &map_callback);
	ros::Subscriber < nav_msgs::Odometry > odomSub ("odom", &odom_callback);
	ros::Subscriber < nav_msgs::MapMetaData > metaSub("map", &metaData_callback);

	nh.subscribe(metaSub);
	//nh.subscribe(odomSub);

	printf("Waiting to receive map data\n");
	
	while(1)
	{
		nh.spinOnce();
		Sleep(100);
	}
	printf("All done!\n");
	return 0;
}
`

romainreignier, 2017-12-07:
Your map is probably too big to fit in the rosserial default buffers (512 Bytes) and may have a size larger than 65 kB (16-bit) which is the max size allowed by the protocol, for 32-bit, see @1r0b1n0 implementation mentioned in this post: https://github.com/ros-drivers/rosserial/issues/337#issuecomment-345557180

wbang8933, 2017-12-10:
I have change the rosserial default buffers to 2MB， and change the max_buffer to 2MB in rosserial_server，What means“the max size allowed by the protocol is 65KB”？ By the way I used it in windows。

romainreignier, 2017-12-10:
The message length is encoded on a 16-bit integer so the protocol cannot
handle longer messages.