ros-drivers/rosserial #347: Missing Ubuntu Package

chrisspen, 2018-01-10:
I'm unable to find rosserial_server package in the Kinetic release on Ubuntu 16. Attempting to run:

    rosrun rosserial_server serial_node _port:=/dev/ttyACM1

returns the error:

    [rospack] Error: package 'rosserial_server' not found

Is rosserial_server not packaged yet?

mikaelarguedas, 2018-01-10:
The package is available for me on Ubuntu xenial:

```
$ apt-cache show ros-kinetic-rosserial-server 
Package: ros-kinetic-rosserial-server
Version: 0.7.7-0xenial-20171129-144254-0800
Architecture: amd64
Maintainer: Mike Purvis <mpurvis@clearpathrobotics.com>
Installed-Size: 1025
Depends: libboost-system1.58.0, libc6 (>= 2.14), libgcc1 (>= 1:3.0), libstdc++6 (>= 5.2), ros-kinetic-roscpp, ros-kinetic-rosserial-msgs, ros-kinetic-rosserial-python, ros-kinetic-std-msgs, ros-kinetic-topic-tools
Priority: extra
Section: misc
Filename: pool/main/r/ros-kinetic-rosserial-server/ros-kinetic-rosserial-server_0.7.7-0xenial-20171129-144254-0800_amd64.deb
Size: 171192
SHA256: 1a50e63d59e952be0d7a563beacbe77b5aa1be40272390ec515514d6d397db0d
SHA1: fb190106a2049663f50890de259551807870365a
MD5sum: 6158e83740371a0a4779c44a6a7535ba
Description: A more performance- and stability-oriented server alternative implemented in C++ to rosserial_python.
Description-md5: 3492172536883772ba44aa4165c13afa
```

Did you try to run:
```
sudo apt update && sudo apt install ros-kinetic-rosserial-server
rospack profile
```
before running:
```rosrun rosserial_server serial_node _port:=/dev/ttyACM1```

chrisspen, 2018-01-10:
Ah, I searched for "rosserial_server", not "rosserial-server". My mistake. Thanks.

closed 2018-01-10 by chrisspen