ros-drivers/rosserial #348: Unable to start rosserial_server: Socket asio error

chrisspen, 2018-01-10:
I'm trying to test the rosserial_server `serial_node` with an Arduino Mega by running:

    ard-reset-arduino /dev/ttyACM1
    rosrun rosserial_server serial_node _port:=/dev/ttyACM1 _baud:=57600

and it's immediately returning the error output:

    [ INFO] [1515595091.755818304]: rosserial_server session configured for /dev/ttyACM1 at 57600bps.
    [ INFO] [1515595091.757117780]: Opened /dev/ttyACM1
    [ WARN] [1515595091.758242777]: Socket asio error, closing socket: asio.misc:2
    [ INFO] [1515595093.758740929]: Opened /dev/ttyACM1
    [ WARN] [1515595093.759567698]: Socket asio error, closing socket: asio.misc:2
    [ INFO] [1515595095.759983976]: Opened /dev/ttyACM1
    [ WARN] [1515595095.760258767]: Socket asio error, closing socket: asio.misc:2
    [ INFO] [1515595097.760723326]: Opened /dev/ttyACM1
    [ WARN] [1515595097.761510876]: Socket asio error, closing socket: asio.misc:2
    [ INFO] [1515595099.761969290]: Opened /dev/ttyACM1
    [ WARN] [1515595099.762223144]: Socket asio error, closing socket: asio.misc:2
    [ INFO] [1515595101.762722703]: Opened /dev/ttyACM1
    [ WARN] [1515595101.762984682]: Socket asio error, closing socket: asio.misc:2
    [ INFO] [1515595103.763453252]: Opened /dev/ttyACM1
    [ WARN] [1515595103.763781272]: Socket asio error, closing socket: asio.misc:2

To confirm the problem wasn't with the Arduino, I also tested it with the legacy rosserial_python `serial_node.py`:

    ard-reset-arduino /dev/ttyACM1
    rosrun rosserial_python serial_node.py _port:=/dev/ttyACM1 _baud:=57600

it works fine and shows the expected output:

    [INFO] [1515595319.613220]: ROS Serial Python Node
    [INFO] [1515595319.636942]: Connecting to /dev/ttyACM1 at 57600 baud
    [INFO] [1515595321.922152]: Note: publish buffer size is 512 bytes
    [INFO] [1515595321.925529]: Setup publisher on battery [sensor_msgs/BatteryState]
    [INFO] [1515595321.945701]: Setup publisher on power/button [std_msgs/Bool]
    [INFO] [1515595321.964321]: Setup publisher on edge/0 [std_msgs/Bool]
    [INFO] [1515595321.983507]: Setup publisher on edge/1 [std_msgs/Bool]
    [INFO] [1515595322.002594]: Setup publisher on edge/2 [std_msgs/Bool]
    [INFO] [1515595322.019815]: Setup publisher on ultrasonic/0 [sensor_msgs/Range]

I read that rosserial_server is experimental, but is there any reason why it shouldn't be able to connect to a basic Arduino Mega?

romainreignier, 2018-01-10:
I am also facing the same issue on Ubuntu 16.04 w86 computers. It was working with 14.04 I think and curiously on ARM processors (Raspberry Pi, Orange Pi...), it is working even with Xenial.
But I have never tried to dig further.
Note that the TCP version of `rosserial_server` works like a charm, so the issue seems to be on the serial part of Boost.asio...

chrisspen, 2018-01-10:
Ah, yes, I should have mentioned I too am running this on an ARM platform (Raspberry Pi 3). This is package only supported on x86?

romainreignier, 2018-01-10:
@chrisspen Ah, it's curious because for me, it works on ARM and not x86.
But I don't think it is related to the architecture but more the OS configuration (like ModemManager using the serial port or bad timeouts...)

PeterQFR, 2018-01-16:
I am running x86 w/ arduino. I uncommented out a few debug lines of a typical run.
I'm a bit unsure how the buffering works but seems to possibly copy too much from the hardware buffer...

```
[ INFO] [1516111293.010958410]: Buffer read of 1 bytes, wi: 152, ri: 111
[ INFO] [1516111293.010994638]: Invoking success callback with buffer of requested size 1 byte(s).
[ INFO] [1516111293.011039362]: Received body of length 1 for message on topic 0.
[ WARN] [1516111293.011075941]: Rejecting message on topicId=0, length=1 with bad checksum. chsum 32
[ INFO] [1516111293.011112510]: Buffer read of 1 bytes, wi: 152, ri: 112
[ INFO] [1516111293.011148731]: Invoking success callback with buffer of requested size 1 byte(s).
[ INFO] [1516111293.011192363]: Buffer read of 1 bytes, wi: 152, ri: 113
[ INFO] [1516111293.011229265]: Invoking success callback with buffer of requested size 1 byte(s).
[ INFO] [1516111293.011272711]: Buffer read of 1 bytes, wi: 152, ri: 114
[ INFO] [1516111293.011308953]: Invoking success callback with buffer of requested size 1 byte(s).
[ INFO] [1516111293.011351627]: Buffer read of 1 bytes, wi: 152, ri: 115
[ INFO] [1516111293.011389740]: Invoking success callback with buffer of requested size 1 byte(s).
[ INFO] [1516111293.011431944]: Buffer read of 1 bytes, wi: 152, ri: 116
[ INFO] [1516111293.011469334]: Invoking success callback with buffer of requested size 1 byte(s).
[ INFO] [1516111293.011512205]: Buffer read of 1 bytes, wi: 152, ri: 117
[ INFO] [1516111293.011548490]: Invoking success callback with buffer of requested size 1 byte(s).
[ INFO] [1516111293.011592675]: Buffer read of 1 bytes, wi: 152, ri: 118
[ INFO] [1516111293.011629049]: Invoking success callback with buffer of requested size 1 byte(s).
[ INFO] [1516111293.011673835]: Buffer read of 1 bytes, wi: 152, ri: 119
[ INFO] [1516111293.011711536]: Invoking success callback with buffer of requested size 1 byte(s).
[ INFO] [1516111293.011753832]: Buffer read of 1 bytes, wi: 152, ri: 120
[ INFO] [1516111293.011790295]: Invoking success callback with buffer of requested size 1 byte(s).
[ INFO] [1516111293.011832560]: Buffer read of 1 bytes, wi: 152, ri: 121
[ INFO] [1516111293.011870064]: Invoking success callback with buffer of requested size 1 byte(s).
[ INFO] [1516111293.011913258]: Buffer read of 1 bytes, wi: 152, ri: 122
[ INFO] [1516111293.011949154]: Invoking success callback with buffer of requested size 1 byte(s).
[ INFO] [1516111293.011992171]: Buffer read of 1 bytes, wi: 152, ri: 123
[ INFO] [1516111293.012028955]: Invoking success callback with buffer of requested size 1 byte(s).
[ INFO] [1516111293.012074729]: Buffer read of 1 bytes, wi: 152, ri: 124
[ INFO] [1516111293.012110962]: Invoking success callback with buffer of requested size 1 byte(s).
[ INFO] [1516111293.012153556]: Buffer read of 1 bytes, wi: 152, ri: 125
[ INFO] [1516111293.012189814]: Invoking success callback with buffer of requested size 1 byte(s).
[ INFO] [1516111293.012231846]: Buffer read of 1 bytes, wi: 152, ri: 126
[ INFO] [1516111293.012271607]: Invoking success callback with buffer of requested size 1 byte(s).
[ INFO] [1516111293.012314265]: Buffer read of 1 bytes, wi: 152, ri: 127
[ INFO] [1516111293.012356388]: Invoking success callback with buffer of requested size 1 byte(s).
[ INFO] [1516111293.012419548]: Buffer read of 1 bytes, wi: 152, ri: 128
[ INFO] [1516111293.012458686]: Invoking success callback with buffer of requested size 1 byte(s).
[ INFO] [1516111293.012542219]: Buffer read of 1 bytes, wi: 152, ri: 129
[ INFO] [1516111293.012583491]: Invoking success callback with buffer of requested size 1 byte(s).
[ INFO] [1516111293.012629047]: Buffer read of 1 bytes, wi: 152, ri: 130
[ INFO] [1516111293.012665459]: Invoking success callback with buffer of requested size 1 byte(s).
[ INFO] [1516111293.012707975]: Buffer read of 1 bytes, wi: 152, ri: 131
[ INFO] [1516111293.012744122]: Invoking success callback with buffer of requested size 1 byte(s).
[ INFO] [1516111293.012785498]: Buffer read of 1 bytes, wi: 152, ri: 132
[ INFO] [1516111293.012822697]: Invoking success callback with buffer of requested size 1 byte(s).
[ INFO] [1516111293.012864450]: Buffer read of 1 bytes, wi: 152, ri: 133
[ INFO] [1516111293.012900537]: Invoking success callback with buffer of requested size 1 byte(s).
[ INFO] [1516111293.012943790]: Buffer read of 1 bytes, wi: 152, ri: 134
[ INFO] [1516111293.012980186]: Invoking success callback with buffer of requested size 1 byte(s).
[ INFO] [1516111293.013021727]: Buffer read of 1 bytes, wi: 152, ri: 135
[ INFO] [1516111293.013057837]: Invoking success callback with buffer of requested size 1 byte(s).
[ INFO] [1516111293.013069235]: Buffer read of 1 bytes, wi: 152, ri: 136
[ INFO] [1516111293.013078747]: Invoking success callback with buffer of requested size 1 byte(s).
[ INFO] [1516111293.013089962]: Buffer read of 1 bytes, wi: 152, ri: 137
[ INFO] [1516111293.013099474]: Invoking success callback with buffer of requested size 1 byte(s).
[ INFO] [1516111293.013110992]: Buffer read of 1 bytes, wi: 152, ri: 138
[ INFO] [1516111293.013120449]: Invoking success callback with buffer of requested size 1 byte(s).
[ INFO] [1516111293.013132020]: Buffer read of 1 bytes, wi: 152, ri: 139
[ INFO] [1516111293.013141462]: Invoking success callback with buffer of requested size 1 byte(s).
[ INFO] [1516111293.013152682]: Buffer read of 1 bytes, wi: 152, ri: 140
[... cut ...]
[ INFO] [1516111295.009686636]: Buffer read of 9 bytes, wi: 16, ri: 7
[ INFO] [1516111295.009745759]: Invoking success callback with buffer of requested size 9 byte(s).
[ INFO] [1516111295.009816883]: Received body of length 9 for message on topic 10.
[ INFO] [1516111295.009936103]: Buffer read of 1 bytes, wi: 0, ri: 0
[ INFO] [1516111295.010002827]: Requesting transfer of at least 1 byte(s).
[ INFO] [1516111295.010096986]: Read operation failed with: asio.misc:2
[ WARN] [1516111295.010161089]: Socket asio error, closing socket: asio.misc:2
```

PeterQFR, 2018-01-17:
Is this error just that in the Async buffer, the buffer is empty, and socket is asked to do an async_read to get at least 1 byte of data from the port as at 

https://github.com/ros-drivers/rosserial/blob/c58e2f11b2ff8060b86a9c50982e31ce02f802ab/rosserial_server/include/rosserial_server/async_read_buffer.h#L94-L100

where transfer_bytes is 1 (this code isn't executed unless it is at least 1. However nothing has been sent from the serial device and the port is empty? This would cause an error.

AlexisTM, 2018-01-17:
From my side; It is not working on x64.

This problem is therefore occurring on all platforms.

Note that it sometimes works.

Also, it always works if I listen to the port via: 

```bash
sudo interceptty -s 'ispeed 57600 ospeed 57600' -l /dev/ttyACM0 /dev/ttyDUMMY | interceptty-nicedump
``` 

And that I start the node on /dev/ttyDUMMY.

PeterQFR, 2018-01-18:
Is this issue [related?](https://answers.ros.org/question/260041/exception-eof-serial-port-boostasiobuffer/)

AlexisTM, 2018-01-18:
Might be...

philboske, 2018-01-19:
Confirmed behaviour of working on 14.04/Indigo and failing on 16.04/Kinetic on ARMHF platform.

Reconnecting the MEGA and trying rosserial_server first gives this behaviour:

```
odroid@devbot_01:/$ roslaunch rosserial_server serial.launch --wait --screen -v
.
.
.
process[rosserial_server-1]: started with pid [32227]
process[rosserial_message_info-2]: started with pid [32228]
[ INFO] [1516320933.261930460]: rosserial_server session configured for /dev/ttyACM0 at 115200bps.
[ INFO] [1516320933.262822506]: Opened /dev/ttyACM0
[INFO] [1516320934.224919]: rosserial message_info_service node
[ INFO] [1516320935.264040260]: Opened /dev/ttyACM0
[ INFO] [1516320937.264956012]: Opened /dev/ttyACM0
[ INFO] [1516320939.265818722]: Opened /dev/ttyACM0
[ INFO] [1516320941.266647805]: Opened /dev/ttyACM0
[ INFO] [1516320943.267631514]: Opened /dev/ttyACM0
[ INFO] [1516320945.268525263]: Opened /dev/ttyACM0
^C[rosserial_message_info-2] killing on exit
[rosserial_server-1] killing on exit
[rosserial_server-1] escalating to SIGTERM
shutting down processing monitor...
... shutting down processing monitor complete
done
```

Rosserial_python works regardless of when the MEGA was connected and what has previously been tried.

```
odroid@devbot_01:/$ rosrun rosserial_python serial_node.py /dev/ttyACM0 _baud:=115200
[INFO] [1516321190.292869]: ROS Serial Python Node
[INFO] [1516321190.305997]: Connecting to /dev/ttyACM0 at 115200 baud
[INFO] [1516321192.432421]: Note: subscribe buffer size is 512 bytes
[INFO] [1516321192.433272]: Setup subscriber on toggle_led [std_msgs/Empty]
```

Rosserial_server produces different behaviour if it is executed after rosserial_python?

```
odroid@devbot_01:/$ roslaunch rosserial_server serial.launch --wait --screen -v
.
.
.
process[rosserial_server-1]: started with pid [32628]
process[rosserial_message_info-2]: started with pid [32629]
[ INFO] [1516321368.340855278]: rosserial_server session configured for /dev/ttyACM0 at 115200bps.
[ INFO] [1516321368.341705157]: Opened /dev/ttyACM0
[ WARN] [1516321368.341995825]: Socket asio error, closing socket: asio.misc:2
[INFO] [1516321369.305647]: rosserial message_info_service node
[ INFO] [1516321370.342624629]: Opened /dev/ttyACM0
[ WARN] [1516321370.342902297]: Socket asio error, closing socket: asio.misc:2
[ INFO] [1516321372.343443599]: Opened /dev/ttyACM0
[ WARN] [1516321372.343750767]: Socket asio error, closing socket: asio.misc:2
^C[rosserial_message_info-2] killing on exit
[rosserial_server-1] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done
```

Hopefully there is a solution looming as we updated our fleet from rosserial_python to rosserial_server in 14.04 to fix stability issues.

AlexisTM, 2018-01-19:
I personally have it working on the ARMhf platform (Odroid XU4) under kinetics and failing often on x64

PeterQFR, 2018-01-19:
If found [this ](https://sourceforge.net/p/asio/mailman/message/3801716/)from a long time ago, but has the same error, and due to the read function returning with zero bytes which was due to a buffer size not being initalise. Not exactly the issue in our case. 
I reimplemented the socket read with the boost::asio::serial_port::read_some(). I get the same error in the same way,
```
 ROS_INFO("mem length, %ld writeIndex %ld HeadRoom %ld", mem_.size(), write_index_, bytesHeadroom());

size_t bytes_read =0;
      boost::system::error_code ec;
      while (bytes_read<transfer_bytes){

    	  int b = stream_.read_some(boost::asio::buffer(&mem_[write_index_], bytesHeadroom()), ec);
    	  if (b == 0)
    	  {
    		  ROS_ERROR("Error");
    		  break;
    	  }
    	  else{
    		  bytes_read+=b;
    		  write_index_+=b;
    	  }
      }
      this->callback(ec, bytes_read);

      /*boost::asio::async_read(stream_,
          boost::asio::buffer(&mem_[write_index_], bytesHeadroom()),
          boost::asio::transfer_at_least(transfer_bytes),
          boost::bind(&AsyncReadBuffer::callback, this,
                      boost::asio::placeholders::error,
                      boost::asio::placeholders::bytes_transferred));*/
```
producing:

[ INFO] [1516362418.129118265]: Starting session.
[ INFO] [1516362418.129210798]: mem length, 1023 writeIndex 0 HeadRoom 1023
[ INFO] [1516362418.129249626]: Successfully read 16 byte(s), now 31 available.
[ INFO] [1516362418.129306963]: Buffer is empty, resetting indexes to the beginning.
[ INFO] [1516362418.129336763]: mem length, 1023 writeIndex 0 HeadRoom 1023
[ERROR] [1516362418.129346557]: Error
[ WARN] [1516362418.129362581]: Socket asio error, closing socket: asio.misc:2

PeterQFR, 2018-01-19:
I did a reimplementation of the async_read, with no effect. This seems to be a boost::asio::error, quite deep in where they interact with software. I was using kernel 4.13 on an x86 machine (which needs at least 4.8 for all peripherals to work)
I downgraded my kernel to 4.4.107 and this specific error goes away, and the message flow is much more consistent and no asio.misc2 errors. 
It seems boost 1.58 was build to interface with earlier kernels and theres been a change which has broken that. I suggest downgrading the kernel or upgrading boost (which is probably not simple with an normal install.. Get out the snaps!)

PeterQFR, 2018-01-19:
actually, I just ran it back up under 4.13 and it is working fine too? I have no idea why but I can speculate that at the kernel/hardware level somthing has changed (probably from changing the kernel) which is allowing it to work.

philboske, 2018-01-20:
@PeterQFR: Going by the output that you provided, the error seems to occur between:

https://github.com/ros-drivers/rosserial/blob/c58e2f11b2ff8060b86a9c50982e31ce02f802ab/rosserial_server/include/rosserial_server/async_read_buffer.h#L151

And:

https://github.com/ros-drivers/rosserial/blob/c58e2f11b2ff8060b86a9c50982e31ce02f802ab/rosserial_server/include/rosserial_server/async_read_buffer.h#L175-L179

The callback that is bound to the async_read() call is stating that there are still bytes available, yet the callSuccessfulCallback() call is reporting that there are no bytes available.

According to the Boost documentation:

`An error code of boost::asio::error::eof indicates that the connection was closed by the peer.`

And this fits in with the behaviour that I posted earlier where it was failing to maintain a connection, even without raising a asio::misc::2 error (which is EOF). Note that I only received aiso::misc::2 errors **after** I had successfully connected previously with rosserial_python. The first attempt at connecting using rosserial_server would hang when issued with a SIGINT (i.e. Ctrl+C), finally being escallated to a SIGTERM unless I disconnected the MEGA.

PeterQFR, 2018-01-21:
Thats different to what I've found. Line 151 wouldn't get executed on error as that callback would return an error code. I also added:
```
else if (error == boost::asio::error::eof)
 {
  ROS_ERROR("EOF Error");
 }
```
in that same callback, and I could confirm that it is returning the error from the boost::asio::async_read(). That why I think it is a hardware configuration error from somewhere in the io_service or at the kernel, as that is where that error code comes from.

philboske, 2018-01-21:
Hmmm, okay, my apologies.

@AlexisTM: Works fine with 'interceptty' redirecting to /dev/ttyDUMMY.

PeterQFR, 2018-01-21:
@philboske  I could be completely wrong though.. I've used boost asio with no problems modifying from the [mavros implementation](https://github.com/mavlink/mavros/blob/master/libmavconn/src/serial.cpp), but have not been able to get serial_server to work. The main difference I can see is that in the mavros implementation is that the io_service is being run in a separate thread. I'll try that here and see if it improves things.

skohlbr, 2018-05-23:
I've run into this issue now, too. Are there any updates regarding this, for instance someone fixing things in a fork and just forgetting to post here? ;)

AlexisTM, 2018-05-23:
I upgraded all ROS packages and the kernel. It never happened again. 

If I remember well, this is due to a Boost version compatibility issue with a certain kernel.

romainreignier, 2018-05-24:
It is the same for me. I have recently tried again and I did not have this issue anymore.

skohlbr, 2018-05-28:
That's interesting I'm on 16.04 with latest kinetic packages from .debs and kernel `4.13.0-41-generic` and the problem persists.

pranavb104, 2019-01-07:
Any solutions to this yet? I am also having the same problems...

philboske, 2019-01-07:
I doubt that it received any attention after the numerous people reporting that a fresh install and update (colloquially 'the solution') seemingly fixed their problems.

I am fairly sure (this is testing my memory) that we implemented rosserial_python on our deployed fleet of teaching robots, and once they become stable, I don't touch them again. Our research fleet is another topic altogether, but we haven't had problems either way.

pranavb104, 2019-01-07:
ok...apparently I just did a fresh install of rosserial server, and now it works fine.

blubbi321, 2021-04-24:
Just had the same error message on multiple machines in multiple configurations (Ubuntu 16/18.04, kinetic vs melodic, different kernels, x86 and arm). Restarting the roscore fixed it for me.

raidhu, 2021-08-11:
[ 62%] Building CXX object rosserial/rosserial_server/CMakeFiles/rosserial_server_serial_node.dir/src/serial_node.cpp.o
In file included from /usr/include/c++/5/cstdint:35:0,
                 from /home/ubuntu/catkin_ws/src/rosserial/rosserial_server/include/rosserial_server/async_read_buffer.h:44,
                 from /home/ubuntu/catkin_ws/src/rosserial/rosserial_server/include/rosserial_server/session.h:50,
                 from /home/ubuntu/catkin_ws/src/rosserial/rosserial_server/include/rosserial_server/serial_session.h:43,
                 from /home/ubuntu/catkin_ws/src/rosserial/rosserial_server/src/serial_node.cpp:40:
/usr/include/c++/5/bits/c++0x_warning.h:32:2: error: #error This file requires compiler and library support for the ISO C++ 2011 standard. This support must be enabled with the -std=c++11 or -std=gnu++11 compiler options.
 #error This file requires compiler and library support \
  ^
In file included from /home/ubuntu/catkin_ws/src/rosserial/rosserial_server/include/rosserial_server/session.h:50:0,
                 from /home/ubuntu/catkin_ws/src/rosserial/rosserial_server/include/rosserial_server/serial_session.h:43,
                 from /home/ubuntu/catkin_ws/src/rosserial/rosserial_server/src/serial_node.cpp:40:
/home/ubuntu/catkin_ws/src/rosserial/rosserial_server/include/rosserial_server/async_read_buffer.h:46:14: error: 'make_signed' in namespace 'std' does not name a template type
 typedef std::make_signed<size_t>::type signed_size_t;
              ^
/home/ubuntu/catkin_ws/src/rosserial/rosserial_server/include/rosserial_server/async_read_buffer.h: In member function 'void rosserial_server::AsyncReadBuffer<AsyncReadStream>::read(size_t, boost::function<void(ros::serialization::IStream&)>)':
/home/ubuntu/catkin_ws/src/rosserial/rosserial_server/include/rosserial_server/async_read_buffer.h:86:5: error: 'signed_size_t' was not declared in this scope
     signed_size_t transfer_bytes = read_requested_bytes_ - bytesAvailable();
     ^
/home/ubuntu/catkin_ws/src/rosserial/rosserial_server/include/rosserial_server/async_read_buffer.h:88:9: error: 'transfer_bytes' was not declared in this scope
     if (transfer_bytes > 0)
         ^
In file included from /home/ubuntu/catkin_ws/src/rosserial/rosserial_server/include/rosserial_server/serial_session.h:43:0,
                 from /home/ubuntu/catkin_ws/src/rosserial/rosserial_server/src/serial_node.cpp:40:
/home/ubuntu/catkin_ws/src/rosserial/rosserial_server/include/rosserial_server/session.h: At global scope:
/home/ubuntu/catkin_ws/src/rosserial/rosserial_server/include/rosserial_server/session.h:532:46: warning: non-static data member initializers only available with -std=c++11 or -std=gnu++11
   int unrecognised_topic_retry_threshold_{ 0 };
                                              ^
/home/ubuntu/catkin_ws/src/rosserial/rosserial_server/include/rosserial_server/session.h:533:31: warning: non-static data member initializers only available with -std=c++11 or -std=gnu++11
   int unrecognised_topics_{ 0 };
                               ^
/home/ubuntu/catkin_ws/src/rosserial/rosserial_server/include/rosserial_server/session.h:532:42: warning: extended initializer lists only available with -std=c++11 or -std=gnu++11
   int unrecognised_topic_retry_threshold_{ 0 };
                                          ^
/home/ubuntu/catkin_ws/src/rosserial/rosserial_server/include/rosserial_server/session.h:532:46: warning: extended initializer lists only available with -std=c++11 or -std=gnu++11
   int unrecognised_topic_retry_threshold_{ 0 };
                                              ^
/home/ubuntu/catkin_ws/src/rosserial/rosserial_server/include/rosserial_server/session.h:533:27: warning: extended initializer lists only available with -std=c++11 or -std=gnu++11
   int unrecognised_topics_{ 0 };
                           ^
/home/ubuntu/catkin_ws/src/rosserial/rosserial_server/include/rosserial_server/session.h:533:31: warning: extended initializer lists only available with -std=c++11 or -std=gnu++11
   int unrecognised_topics_{ 0 };
                               ^
rosserial/rosserial_server/CMakeFiles/rosserial_server_serial_node.dir/build.make:62: recipe for target 'rosserial/rosserial_server/CMakeFiles/rosserial_server_serial_node.dir/src/serial_node.cpp.o' failed
make[2]: *** [rosserial/rosserial_server/CMakeFiles/rosserial_server_serial_node.dir/src/serial_node.cpp.o] Error 1
CMakeFiles/Makefile2:4849: recipe for target 'rosserial/rosserial_server/CMakeFiles/rosserial_server_serial_node.dir/all' failed
make[1]: *** [rosserial/rosserial_server/CMakeFiles/rosserial_server_serial_node.dir/all] Error 2
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j1" failed
ubuntu@ubiquityrobot:~/catkin_ws$ rosversion rosserial_server
0.9.2
ubuntu@ubiquityrobot:~/catkin_ws$ cat /usr/include/boost/version.hpp | grep "BOOST_LIB_VERSION"
//  BOOST_LIB_VERSION must be defined to be the same as BOOST_VERSION
#define BOOST_LIB_VERSION "1_58"

**I sourced the rosserial driver _ws and do catkin_make it shows this error
im using ubuntu16.04, ros kinetic
this error caused due to rosserial_server lilbrary codes how to solve this
but the rosserial works fine im able to communicate with arduino**

AlexisTM, 2021-08-11:
From the looks of it, you need to enable C++11 for your project.