ros-drivers/rosserial #351: Publisher stops responding, unable to flush

chrisspen, 2018-01-16:
I've been trying to diagnose sync errors between an Arduino Mega and a serial_node.py.

I created a simple string publisher, to publish generic diagnostic messages, thinking that I was having memory fragmentation issues and that the built diagnostic messages were too big, contributing to this problem.

I define my publisher like:

    ros::Publisher diagnostics_publisher = ros::Publisher("diagnostics_relay", &string_msg);

My setup() looks like:

    nh.getHardware()->setBaud(57600);
    nh.initNode();
    nh.advertise(diagnostics_publisher);

My loop() looks like:

    if (millis() - last_memory_report >= 10000) { // 300000=5 minutes
        last_memory_report = millis();

        snprintf(buffer, MAX_OUT_CHARS, "freeMemory(): %d", freeMemory());
        nh.loginfo(buffer);
    }

    report_diagnostics = false;
    if (millis() - last_diagnostic >= 10000) {
        last_diagnostic = millis();
        report_diagnostics = true;
    }

    if (report_diagnostics) {
        snprintf(buffer, MAX_OUT_CHARS, "somekey:somevalue");
        string_msg.data = buffer;
        diagnostics_publisher.publish(&string_msg);
    }

When I launch serial_node.py, I receive a ton of messages from the "diagnostics_relay" topic on my Arduino, but after a minute, they suddenly all stop. Running `rostopic echo /arduino/diagnostics_relay` shows many messages at first, and then nothing. Killing and relaunching rostopic has no effect. However, the Arduino does not lose sync.

At first I thought my Arduino had crashed, but I found I can still call services on it, and it still responds to other topics that it subscribes to. Additionally, I installed the MemoryFree library, to help determine if I was simply running out of memory, and logged this value in my loop. However, even after the Arduino stops publishing diagnostics, I still see these memory logging messages, implying the serial_node.py still has a working connection. It shows the free memory between the stack and heap never drops below 9013.

Why is my custom publisher appearing to lock up? Since the Arduino is still responding to other ROS messages, is it fair to rule out the firmware? What else could be causing this? I am sending a ton of diagnostic messages. Are there outgoing buffer limits on the Arduino? I tried adding Serial.flush() on the Arduino after my publish() calls, but that had no effect. How can I verify the Arduino's ROS transmit buffer is empty?

chrisspen, 2018-01-16:
I just realized that SerialClient is not threaded. Do you think it's possible that there's a deadlock happening because I'm sending too much data from the Arduino, causing the serial writes to block until the transmit buffer is empty, but serial_node.py isn't reading, because its single main thread is trying to write to the Arduino, which isn't responding because it's also waiting to transmit?

mikepurvis, 2018-01-16:
Do you see the same thing when running this firmware against `rosserial_server`?

chrisspen, 2018-01-16:
I was unable to get `rosserial_server` to run on my ARM platform, as I documented in [issue 348](https://github.com/ros-drivers/rosserial/issues/348).

chrisspen, 2018-01-17:
I've rewritten SerialClient to use a thread for writing and that seems to have resolved all my reliability problems. I need to do a lot of testing, but if I'm right, this could drastically make `serial_node.py` more reliable and fix a lot of those fatal sync errors users report. I'll update my PR when I've cleaned it up.

shuhaowu, 2018-08-09:
Can you share your implementation of the threaded serial node?

chrisspen, 2018-08-10:
I did. It's already been merged.

closed 2018-08-10 by chrisspen

thuhoang-bk, 2022-04-06:
A bit late but my source of problem was the power supply to Mega board, i used battery and usb port at the same time. When i unplugged the battery, problem solved.