ros-drivers/rosserial #353: ROS Nodemcu Arduino compilation issue

chrissunny94, 2018-01-18:
Hey guys , i am trying to run this https://github.com/agnunez/espros

I got an error like this when i tried to compile ,
`
'class ArduinoHardware' has no member named 'setConnection'
 nh.getHardware()->setConnection(server, serverPort);
                 ^
exit status 1
'class ArduinoHardware' has no member named 'setConnection'
`
Any one faced similar problems before ?

I am trying to run this on Ros-Indigo (Might that be the problem , should i try some newer version)

https://www.youtube.com/watch?v=o_gFG... , this is what i am trying to do right now .

I can control the bot vai UDP packets generated from a python script .

But i would like to get it ROS enabled right from the MCU level . I know i can write a middle ware python script to make it ROS enabled .(but i want a cleaner code)

romainreignier, 2018-01-18:
Why are you using this external repo? The ESP8266 has been included in this git repo. But it might not be available in ROS Indigo Ubuntu packages, so you have to clone this repo in your Catkin workspace, build it, source it and then re-generate the `ros_lib` library in your Arduino Libraries folder.

Because the `setConnection()` method exists in the current version of `rosserial_arduino`, see here:
https://github.com/ros-drivers/rosserial/blob/c58e2f11b2ff8060b86a9c50982e31ce02f802ab/rosserial_arduino/src/ros_lib/ArduinoTcpHardware.h#L54
So you should not get that error.

R0dri, 2018-06-14:
@romainreignier thank you, had the same problem and it was because I was not using this git repo as you stated. Should this be closed?

arom25, 2022-01-10:
Hi @chrissunny94, make sure you add #define **ROSSERIAL_ARDUINO_TCP** before including the file "ros.h".

Like this:
#define ROSSERIAL_ARDUINO_TCP
#include "ros.h"

Thanks!

arom25, 2022-01-10:
Hi @chrissunny94, make sure you add #define ROSSERIAL_ARDUINO_TCP before including the file "ros.h".

Like this:
#define ROSSERIAL_ARDUINO_TCP
#include "ros.h"

Thanks!

arom25, 2022-01-10:
Hi @chrissunny94, make sure you add **#define ROSSERIAL_ARDUINO_TCP** before including the file "ros.h".

Like this:
```
#define ROSSERIAL_ARDUINO_TCP
#include "ros.h"
IPAddress server(192, 168, 0, 100);
const uint16_t serverPort = 11411;
nh.getHardware()->setConnection(server, serverPort);
```

Thanks!

noshluk2, 2022-01-14:
> 

I am facing the same issue of restarting when ever i add "#define ROSSERIAL_ARDUINO_TCP"

sajidmohd717, 2022-04-27:
> Hi @chrissunny94, make sure you add **#define ROSSERIAL_ARDUINO_TCP** before including the file "ros.h".
>
> Like this:
>
> ```
> #define ROSSERIAL_ARDUINO_TCP
> #include "ros.h"
> IPAddress server(192, 168, 0, [...]

Thanks!, this solved it for me