ros-drivers/rosserial #354: "Mismatched protocol version in packet" when building from source in ROS kinetic

AgtCain, 2018-01-26:
I need to build rosserial for kinetic from source as I want to implement my own custom package, which includes a custom message (more to come in the future). Prebuilt rosserial libraries for kinetic work well and I am able to establish connection. 

However, I tried building rosserial from source having the extra package. It builds successfully but when I run rosserial_python, it gives me the following:

`
[ERROR] [1516956433.836973]: Mismatched protocol version in packet: lost sync or rosserial_python is from different ros release than the rosserial client
[INFO] [1516956433.837802]: Protocol version of client is unrecognized, expected Rev 1 (rosserial 0.5+)
`

The exact same code from the arduino side works perfectly with the prebuilt libraries. I even tried not using my custom message and use a tf message without success.

Any ideas?

AgtCain, 2018-01-26:
A small update on this:
I am using ESP32 for my tests. I noticed that I couldn't call setBaud() either, so I changed the #include ArduinoTcpHardware to #include ArduinoHardware. Sure enough, setBaud() is recognized and the connection is established successfully.

That was on the d288a48c3fa22a417b0ae227dd823f12e8c59c9a commit when ESP32 support was added. Now I will try on the latest commit, but that should be the issue.

Plus, having 512B buffers on the ESP32 is quite conservative. Maybe they should be increased.

romainreignier, 2018-01-26:
Are you using the WiFi connection or a serial port?
Because there is no baudrate in a TCP connection (`ArduinoTcpHardware`).

AgtCain, 2018-01-26:
I want to use the serial port. Is there a flag somewhere to trigger this? Because at the moment, ArduinoTcpHardware seems to be replacing rather than extending ArduinoHardware.

romainreignier, 2018-01-26:
Yes exactly, sadly, using an ESP8266 or ESP32, the TCP version is automatically chosen, see here:

https://github.com/ros-drivers/rosserial/blob/c58e2f11b2ff8060b86a9c50982e31ce02f802ab/rosserial_arduino/src/ros_lib/ros.h#L40-L44

One should add a new `#define` to for using the classic `ArduinoHardware`.

But as @mikepurvis already mentioned several times, there is already too much `defines` in that code, we should try to find a cleaner way to do all theses selections.

But I think that for now, a new `#define` could be added.

AgtCain, 2018-01-29:
Ok, fair enough!

isircu, 2020-02-09:
Sorry to bump an old issue but I experience something similar.  
Trying to get the basic HelloWorld to work with an ESP32 board (Adafruit ESP32 Feather) connected by USB. 
If I upload the code to an Arduino it works as expected but for the ESP32 I get the `protocol mismatch` error.  Changing to `#include ArduinoHardware.` as @AgtCain suggested makes Rosserial hang a while at `Requesting topics` and then it throws:
`Unable to sync with device; possible link problem or link software version mismatch such as hydro rosserial_python with groovy Arduino`

Russ76, 2021-04-17:
I got this same error message from a Teensy board that was programmed from a laptop with Ubuntu 18.04 and Melodic. The other code that was running on the robot used Ubuntu 20.04 and Noetic. I thought the versions should be fine but Rosserial worked for only 10 seconds before throwing the error and quitting.

Russ76, 2021-04-19:
This message comes up from Rosserial when there is a problem with the communications. It may be reporting a performance issue and not actually a version mismatch. It's just a general error message, actually. Try slowing down the communications (using less bandwidth) and it may straighten out. It did for me.

tsadarsh, 2022-06-11:
This issue still persists. I am running ROS Noetic on Ubuntu 20.04

joedavid-tuni, 2023-08-04:
My ESP-32 was running at 240Mhz, when I changed it to 160Mhz this error was gone. You change it in the Arduino IDE tools>CPU frequency

Marcus305, 2023-11-24:
I have the same error with my ESP32 WROOM. I make as the @joedavid-tuni said and I changed to 160Mhz and work it!

furquansh, 2024-01-18:
@HerickDallagnol @Trushal23 @tsadarsh  Did yall manage to solve the error , if yes please do share

HerickDallagnol, 2024-01-18:
The only way I managed to make it work was by resetting the ESP32 during the connection with rosserial. After running and encountering this error, I hold the reset button on the ESP, and magically it works. I've tried several other methods, but this was the only one that worked in my case. I'm exploring possibilities to reset the ESP32 programmatically without the need to press the button

HerickDallagnol, 2024-01-18:
> @HerickDallagnol @Trushal23 @tsadarsh  Did yall manage to solve the error , if yes please do share 
> 
> 
> 

The only way I managed to make it work was by resetting the ESP32 during the connection with rosserial. After running and encountering this error, I hold the reset button on the ESP, and magically it works. I've tried several other methods, but this was the only one that worked in my case. I'm exploring possibilities to reset the ESP32 programmatically without the need to press the button

Mio-Atse, 2025-02-11:
Thank all of you for your hardwork,

I have some specific problem on this topic. I will provide all the information in below. Basically, i am trying to subscribe the rostopics that run on my Arduino Uno with SimulIDE (yeah i know it is crazy but it is fit for me rn.) i can see handshake on the Arduino rx tx serial terminal as. but i cant sub to rostopics and use that i flash.  a bit complicated process but i think you folks could understand what i am trying to do. Here we go in detail:

Observed Behavior
Once the connection is established, I receive the following serial input reading in SimulIDE:
`255 254 0 0 255 0 0 255 255 254 0 0 255 0 0 255 255 254 0 0 255 0 0 255 255 254 0 0 255 0 0 255`
The LED does not blink as expected. And i can pass messages using echo screen between to end points and see them with screen also.

When i run the following `rosnode list `command, i see just this

```
/rosout
/serial_node
```

and also for `rostopic list`

`
/diagnostics
/rosout
/rosout_agg
`

I can confirm the connection is established because I see connection logs in the socat terminal.
System Configuration & Connections
ROS Serial Connection (macOS)

I create a virtual connection with socat :

`socat -d -d pty,raw,echo=0,link=/tmp/ttyv0 pty,raw,echo=0,link=/tmp/ttyv1`

ROS serial node is started with:
`rosrun rosserial_python serial_node.py /tmp/ttyv1 --b 57600`
SimulIDE’s serial terminal is connected to /`tmp/ttyv0` with 57600 baud rate.
These connections are managed via socat, which links virtual serial ports (ttys). 

ROS Serial Terminal Output (with errors)
Despite seeing handshake data from RX/TX, I get the following error:

```
[ERROR] [1739277146.825620]: Unable to sync with device; possible link problem or link software version mismatch such as hydro rosserial_python with groovy Arduino  
[INFO] [1739277146.828098]: Requesting topics...
```

Code Implementation
```
Arduino Code (Flashed to the Board)
/* 
 * rosserial Subscriber Example
 * Blinks an LED on callback
 */

#include <ros.h>
#include <std_msgs/Empty.h>

ros::NodeHandle nh;

void messageCb( const std_msgs::Empty& toggle_msg ){
  // Toggle the LED by writing HIGH - current state
  digitalWrite(LED_BUILTIN, HIGH - digitalRead(LED_BUILTIN));
}

ros::Subscriber<std_msgs::Empty> sub("toggle_led", &messageCb);

void setup() { 
  pinMode(LED_BUILTIN, OUTPUT);
  nh.initNode();
  nh.subscribe(sub);
}

void loop() {  
  nh.spinOnce();
  delay(1);
}
```