ros-drivers/rosserial #358: Error in installation Debian 9 (Stretch) with ROS Lunar

dantuluri, 2018-03-04:
I've been trying to install and it always get stuck at
`[ 95%] Building CXX object rosserial/rosserial_server/CMakeFiles/rosserial_server_serial_node.dir/src/serial_node.cpp.o
`
I created the catkin workspace by:
1. mkdir -p catkin-ws/src
2. catkin make
3. sourcing it
4. cloning this repo into src and then going back to catkin-ws to make it again

How should I go about this?

TuxLeon, 2018-03-04:
Depending of your target you should go on with one of the links the Client Libraries chapter on this site.
http://wiki.ros.org/rosserial#Client_Libraries

dantuluri, 2018-03-07:
Restarted task multiple times. Fixed the trivial issue.

closed 2018-03-07 by dantuluri

welkerCode, 2018-08-01:
@dantuluri I am having the same problem.  I get stuck at 95%, and am getting a lot of cpp output before the build fails.  Restarting catkin_make over and over eventually allows me to finish the build, but now I am getting an "ImportError: cannot import name Serial" every time I try to run rosrun rosserial_arduino /dev/ttyUSB0.   Did you have a similar problem?