ros-drivers/rosserial #365: RosSerial and arduino communication problem

shwetankv007, 2018-03-30:
Hello everyone,
I have been searching a lot about this issue however found no reliable solution yet, i am using ROS kinetic on an ubuntu machine (16.04 LTS), i am trying to communicate with the Arduino board, i added the required libraries to arduino IDE as well as installed the ROS packages (rosserial metapackage). 

The package works just fine with the examples provided in arduino library, i tried using the SERVO example and PUBSUB example as well, both of them worked fine but when i tried to use it on my custom code that subscribes to joint states data of a robotic arm and move the servos accordingly communication error persists:-

`rosrun rosserial_python serial_node.py _port:=/dev/ttyACM1
[INFO] [1522413477.241087]: ROS Serial Python Node
[INFO] [1522413477.245848]: Connecting to /dev/ttyACM1 at 57600 baud
[INFO] [1522413479.395312]: Note: subscribe buffer size is 1024 bytes
[INFO] [1522413479.395781]: Setup subscriber on joint_states [sensor_msgs/JointState]
Traceback (most recent call last):
  File "/opt/ros/kinetic/lib/rosserial_python/serial_node.py", line 89, in <module>
    client.run()
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rosserial_python/SerialClient.py", line 515, in run
    self.requestTopics()
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rosserial_python/SerialClient.py", line 397, in requestTopics
    self.port.flushInput()
  File "/usr/lib/python2.7/dist-packages/serial/serialutil.py", line 532, in flushInput
    self.reset_input_buffer()
  File "/usr/lib/python2.7/dist-packages/serial/serialposix.py", line 566, in reset_input_buffer
    termios.tcflush(self.fd, termios.TCIFLUSH)
termios.error: (5, 'Input/output error')
`
I tried to increase the default buffer size and increased it to 1024 for arduino Mega but nothing changes.
the arduino code that i am using to subscribe to the joint_state data is:-
`#include <Servo.h>
#include <ros.h>
#include <sensor_msgs/JointState.h>

Servo servo1;
Servo servo2;
Servo servo3;
Servo servo4;

double radiansToDegrees(float position_radians)
{
  return position_radians * 57.3248;
}

ros::NodeHandle nh;
int base=90;
int torso_joint=90;
int elbow=90;
int wrist=90;

ros::Subscriber<sensor_msgs::JointState> sub("joint_states", servo_cb);

void setup()
{
  servo1.attach(9);
  servo2.attach(10);
  servo3.attach(11);
  servo4.attach(12);
  nh.getHardware()->setBaud(57600);
  nh.initNode();
  nh.subscribe(sub);  

}

void loop()
{
nh.spinOnce();
servo1.write(base);
servo2.write(torso_joint);
servo3.write(elbow);
servo4.write(wrist);
delay(10);
}

void servo_cb(const sensor_msgs::JointState& cmd_msg)
{
//servo1.write(radiansToDegrees(cmd_msg.position[0]));
//servo2.write(radiansToDegrees(cmd_msg.position[1]));
//servo3.write(radiansToDegrees(cmd_msg.position[2]));
//servo4.write(radiansToDegrees(cmd_msg.position[3]));
base = radiansToDegrees(cmd_msg.position[0]);
torso_joint = radiansToDegrees(cmd_msg.position[1]);
elbow = radiansToDegrees(cmd_msg.position[2]);
wrist = radiansToDegrees(cmd_msg.position[3]);
}
`

I have searched previous issue relating to Rosserial and arduino communication but none seems to help out resolve my problem. I also tried to use the rosserial_server package, it didn't show any topics published and while inspecting the /rosout topic i realized that it keeps on making connection request but no reply is received.

`header: 
  seq: 15
  stamp: 
    secs: 1522412235
    nsecs: 187883450
  frame_id: ''
level: 2
name: "/rosserial_server_serial_node"
msg: "Opened /dev/ttyACM0"
file: "/tmp/binarydeb/ros-kinetic-rosserial-server-0.7.7/include/rosserial_server/serial_session.h"
function: "SerialSession::attempt_connection"
line: 92
topics: [/rosout]
---
header: 
  seq: 16
  stamp: 
    secs: 1522412237
    nsecs: 188443927
  frame_id: ''
level: 2
name: "/rosserial_server_serial_node"
msg: "Opened /dev/ttyACM0"
file: "/tmp/binarydeb/ros-kinetic-rosserial-server-0.7.7/include/rosserial_server/serial_session.h"
function: "SerialSession::attempt_connection"
line: 92
topics: [/rosout]
---
header: 
  seq: 17
  stamp: 
    secs: 1522412239
    nsecs: 189120522
  frame_id: ''
level: 2
name: "/rosserial_server_serial_node"
msg: "Opened /dev/ttyACM0"
file: "/tmp/binarydeb/ros-kinetic-rosserial-server-0.7.7/include/rosserial_server/serial_session.h"
function: "SerialSession::attempt_connection"
line: 92
topics: [/rosout]
`

Any suggestions would be appreciated.
thank you.

Baycken, 2018-05-19:
Can you post a sample message you are trying to send via `rostopic echo`?

7ayushgupta, 2019-02-05:
Are you using the Arduino to power the servos? We usually faced the same problem when we did that. Did you solve it?

zoubata, 2019-05-19:
Hi,
I have the same issue.
on rosserial server, on The log I see some error on the serial access, ros disconnect and restart, but the restart isn't every robust, some topic are dropped. 
did you use an raspberry pi as host?
Dis you supply you servo from arduino board supplied by the USB of the raspi?
Try tu use external supply, I suspect, on raspi, power issue creating i/o errors.