ros-drivers/rosserial #370: Document how-to for use of rosserial_arduino with Due and CMake-based build

skohlbr, 2018-05-14:
Already asked about this on [ROS Answers](https://answers.ros.org/question/290333/how-to-use-a-arduino-due-in-a-cmake-based-package-using-rosserial-arduino/) to no avail. There's also a [previous Q/A](https://answers.ros.org/question/225519/arduino-due-cmake-with-rosserial_arduino/) quite some time ago that I found only just now with no working answer. I really would like to avoid going through the Arduino IDE for building+flashing, but it's unclear to me how this works for the Due.

mikepurvis, 2018-05-14:
The arduino-cmake project never supported non-AVR boards, see https://github.com/queezythegreat/arduino-cmake/issues/159.

The next step on this would be to investigate the effort required to move rosserial_arduino over to this fork, which I believe supports Due and other "modern" Arduinos: https://github.com/arduino-cmake/arduino-cmake

skohlbr, 2018-05-14:
Not what I wanted to read, but makes perfect sense :) . Wasn't aware that arduino-cmake is a external dependency pulled in. Sounds like integrating the fork would be great, unfortunately don't have the cycles to volunteer for that.