ros-drivers/rosserial #375: Support for Arduino MKR1000

STGRobotics, 2018-06-10:
I'm currently working on a project that uses rosserial to communicate with a Arduino MKR1000. Can you assist.

Thank you

romainreignier, 2018-06-11:
Are you using the standard serial port (Serial.begin()...) ?

Can you describe what is not working with the current version of rosserial?

STGRobotics, 2018-06-12:
I'm using the HelloWorld example, getting this when compiling

WARNING: library Rosserial_Arduino_Library claims to run on (avr) architecture(s) and may be incompatible with your current board which runs on (samd) architecture(s).

romainreignier, 2018-06-12:
Oh, I have never seen this warning before. Must be a recent feature. Which
version of the Arduino IDE are you using?

Anyway, if it is just a warning, does the program compile and works?

STGRobotics, 2018-06-12:
Arduino IDE 1.8.5, Compiled and updated flush. have not tested.

WARNING: library Rosserial_Arduino_Library claims to run on (avr) architecture(s) and may be incompatible with your current board which runs on (samd) architecture(s).
Sketch uses 13672 bytes (5%) of program storage space. Maximum is 262144 bytes.
Atmel SMART device 0x10010005 found
Device       : ATSAMD21G18A
Chip ID      : 10010005
Version      : v2.0 [Arduino:XYZ] Dec 20 2016 15:36:43
Address      : 8192
Pages        : 3968
Page Size    : 64 bytes
Total Size   : 248KB
Planes       : 1
Lock Regions : 16
Locked       : none
Security     : false
Boot Flash   : true
BOD          : true
BOR          : true
Arduino      : FAST_CHIP_ERASE
Arduino      : FAST_MULTI_PAGE_WRITE
Arduino      : CAN_CHECKSUM_MEMORY_BUFFER
Erase flash
done in 0.784 seconds

Write 13960 bytes to flash (219 pages)

[========                      ] 29% (64/219 pages)
[=================             ] 58% (128/219 pages)
[==========================    ] 87% (192/219 pages)
[==============================] 100% (219/219 pages)
done in 0.094 seconds

Verify 13960 bytes of flash with checksum.
Verify successful
done in 0.014 seconds
CPU reset.

romainreignier, 2018-06-12:
OK, so you should try and report if it is workinging.

The warning is shown because the library does not provide a
`library.properties` file like describe [here](
https://github.com/arduino/arduino/wiki/arduino-ide-1.5:-library-specification
).
We should add one to avoid the warning.

Feel free to propose a Pull Request.

mikepurvis, 2018-06-12:
Please see https://github.com/ros-drivers/rosserial/issues/370#issuecomment-388936749 for the next steps on supporting ARM-based arduinos.

closed 2018-06-12 by mikepurvis

safariace, 2020-11-18:
@mikepurvis Did this ever get resolved.  Are ARM based arduinos now supported?