ros-drivers/rosserial #377: "avr/pgmspace.h: No such file or directory" on ESP8266.

R0dri, 2018-06-13:
I am using this library together with my NodeMCU and other ESP8266-12 modules.
But the library/compiler throws the following error: "fatal error: avr/pgmspace.h: No such file or directory".

I found the following solution:

at ros.h you have to replace

`#include <avr\pgmspace.h>`

with

>#if (defined(__AVR__))
>#include <avr\pgmspace.h>
>#else
>#include <pgmspace.h>
>#endif

This works fine for me.

This solution was suggested by: [https://github.com/SodaqMoja/Sodaq_DS3231/issues/5](url) having a similar issue with another library. Should this be a merge request?

romainreignier, 2018-06-13:
Hello,

I do not think this issue is related with rosserial because I cannot see
mention of `avr/pgmspace.h` in the current [ros.h](
https://github.com/ros-drivers/rosserial/blob/fe1e233f39d56931b5e223c5e53a2d9c2f9fb664/rosserial_arduino/src/ros_lib/ros.h)
file.

Furthermore we already used ESP8266 board with rosserial without facing
this issue.

Please, make sure you have a recent version of `rosserial_arduino` (maybe
you need to use the git version instead of the one provided by your
distribution if it is too old.

R0dri, 2018-06-14:
I'm sorry, you where right. I tried to use the libraries imported by the Arduino IDE library manager. Just downloaded and compiled from source and it worked flawlessly.

closed 2018-06-14 by R0dri