ros-drivers/rosserial #384: Melodic release please

doisyg, 2018-07-19:
Support for the current LTS would be nice!

CanyonTurtle, 2018-07-24:
@doisyg, I am using it on Melodic by building from source, it does seem to work. I have used rosserial_arduino successfully.

welkerCode, 2018-08-01:
@CanyonTurtle, I tried installing rosserial from source, and after adding it to my workspace and trying to run: 

rosrun rosserial_arduino serial_node.py /dev/ttyUSB0

I get the following error:

ImportError: cannot import name Serial.

Any tips or tricks on what I am missing?  I tried to install all dependencies in my workspace using rosdep, and tried installing python's serial package, but I have been stuck on this.

joe-123, 2018-08-02:
@CanyonTurtle 
I am quite new to ROS. Can you give instructions how to build from source? Then I could test it too. A link with instructions should be enough.

CanyonTurtle, 2018-08-02:
@joe-123 To build from source, make your own workspace for rosserial and install using `catkin_make`:

```bash
source /opt/ros/melodic/setup.bash
mkdir -p ~/workspace/src; cd ~/workspace/src
git clone https://github.com/ros-drivers/rosserial.git
cd ..; catkin_make; catkin_make install
```

This will create a new rosserial workspace folder for you called `workspace`, and it will then download the source files and install the scripts, such as `rosserial_arduino serial_node.py` . To run the scripts, open a new terminal window and run `source ~/worspace/install/setup.bash` (this is referred to as sourcing the workspace). Now you should be able to run the `serial_node` script. @joe-123  hopefully this gives you an idea of what to do next. Also, you will need to install system dependencies (which is seperate from 
building, see the next part of this comment for help with this).

 @welkerCode I am not sure exactly how to help, what version of ROS/ubuntu are you on (melodic, I presume)? It seems that you are missing system dependencies. Have you tried this? I think I remember using this one instead of just `rosdep install`.

"
### Install dependency of all packages in the workspace
This usecase shows even more powerful feature of rosdep.

Go to the top directory of your catkin workspace where the source code of the ROS packages you'd like to use are. Then run:

```bash
rosdep install --from-paths src --ignore-src -r -y
```
This command magically installs all the packages that the packages in your catkin workspace depend upon but are missing on your computer.
" - http://wiki.ros.org/rosdep

Hope this helps!

welkerCode, 2018-08-02:
@CanyonTurtle Thank you for your quick reply!  I tried reinstalling the repository like you instructed.  In the past, I hadn't used the "catkin_make install" at the end, so I thought that might be the problem.   I also tried using your rosdep command, but I got the message

`ubuntu@ubuntu:~/catkin_ws$ rosdep install --from-paths src --ignore-src -r -y
#All required rosdeps installed successfully
`
 However, I am still getting the following error code when I try to run:

`rosrun rosserial_arduino serial_node.py /dev/ttyUSB0'

`Traceback (most recent call last):
  File "/home/ubuntu/catkin_ws/install/share/rosserial_arduino/serial_node.py", line 37, in <module>
    from rosserial_arduino import SerialClient
  File "/home/ubuntu/catkin_ws/install/lib/python2.7/dist-packages/rosserial_arduino/__init__.py", line 1, in <module>
    from SerialClient import *
  File "/home/ubuntu/catkin_ws/install/lib/python2.7/dist-packages/rosserial_arduino/SerialClient.py", line 41, in <module>
    from serial import Serial
ImportError: cannot import name Serial
ubuntu@ubuntu:~/catkin_ws$ ./src/rosserial/rosserial_arduino/nodes/serial_node.py
Traceback (most recent call last):
  File "./src/rosserial/rosserial_arduino/nodes/serial_node.py", line 37, in <module>
    from rosserial_arduino import SerialClient
  File "/home/ubuntu/catkin_ws/install/lib/python2.7/dist-packages/rosserial_arduino/__init__.py", line 1, in <module>
    from SerialClient import *
  File "/home/ubuntu/catkin_ws/install/lib/python2.7/dist-packages/rosserial_arduino/SerialClient.py", line 41, in <module>
    from serial import Serial
ImportError: cannot import name Serial
`
I am running Ubuntu server 18.04 on a Raspberry Pi B.  I am also using Bare Bones Ros Melodic.

I am thinking the problem is coming while I try to run the catkin_make command.  It always seems to get stuck at 95% while building, and I have to kill the program and restart it like 10 times before it finishes.  And when it does, it spits out output like the following:

`/usr/include/c++/7/bits/stl_uninitialized.h:115:5: note: parameter passing for ar   gument of type ‘std::move_iterator<boost::asio::detail::timer_queue<boost::asio::   detail::forwarding_posix_time_traits>::heap_entry*>’ changed in GCC 7.1
     uninitialized_copy(_InputIterator __first, _InputIterator __last,
     ^~~~~~~~~~~~~~~~~~
/usr/include/c++/7/bits/stl_uninitialized.h:115:5: note: parameter passing for ar   gument of type ‘std::move_iterator<boost::asio::detail::timer_queue<boost::asio::   detail::forwarding_posix_time_traits>::heap_entry*>’ changed in GCC 7.1
/usr/include/c++/7/bits/stl_uninitialized.h:134:41: note: parameter passing for a   rgument of type ‘std::move_iterator<boost::asio::detail::timer_queue<boost::asio:   :detail::forwarding_posix_time_traits>::heap_entry*>’ changed in GCC 7.1
  __uninit_copy(__first, __last, __result);
                                         ^
In file included from /usr/include/c++/7/memory:65:0,
                 from /usr/include/boost/asio/detail/addressof.hpp:21,
                 from /usr/include/boost/asio/detail/handler_alloc_helpers.hpp:1    ,
                 from /usr/include/boost/asio/detail/bind_handler.hpp:19,
                 from /usr/include/boost/asio/detail/wrapped_handler.hpp:18,
                 from /usr/include/boost/asio/io_service.hpp:24,
                 from /usr/include/boost/asio/basic_io_object.hpp:19,
                 from /usr/include/boost/asio/basic_socket.hpp:20,
                 from /usr/include/boost/asio/basic_datagram_socket.hpp:20,
                 from /usr/include/boost/asio.hpp:21,
                 from /home/ubuntu/catkin_ws/src/rosserial/rosserial_server/src/u   dp_socket_node.cpp:34:
`

I saw a similar issue at https://github.com/ros-drivers/rosserial/issues/358 (same repo, closed issue, back in March of this year).  They also got stuck at 95% like me, but after running catkin_make over and over they got theirs to work.  Mine still has that import problem.

I know I am doing something wrong.  If you have any tips, I'd appreciate it!  Thank you!

CanyonTurtle, 2018-08-02:
Oh, @welkerCode I know what you are talking about! I actually spent an hour or so trying to get rosserial running on a raspberry pi 2 about 2 weeks ago, on LXDE 16.04 with ROS kinetic. It got stuck at 95% and the error was just like you described. I never investigated how to fix it, I gave it up and worked on something else. So, unfortunately, I don't know the fix. Hopefully someone else can provide a suggestion for how to fix this - it sounds like lots of people will encounter it.

Best of luck!

welkerCode, 2018-08-02:
@CanyonTurtle Thanks for your input!  I'm glad its not just me.  Makes me feel a little better.  I will let you know if I find a fix.

CanyonTurtle, 2018-08-02:
https://answers.ros.org/question/249515/raspberry-pi-crashes-when-i-try-to-build-the-catkin-workspace/

@welkerCode maybe this will help?

CanyonTurtle, 2018-08-02:
I agree - good point. For Serial, have you run `pip install seria`l / `pip3 install serial`?

Do you need to run rosdep init?

did you follow the instructions for initializing rosdep? http://wiki.ros.org/rosdep

welkerCode, 2018-08-02:
I am pretty sure that I correctly initialized rosdep:  

`ubuntu@ubuntu:~/catkin_ws$ rosdep init
ERROR: default sources list file already exists:
        /etc/ros/rosdep/sources.list.d/20-default.list
Please delete if you wish to re-initialize
`
However, I have just installed the desktop version of ROS (not the full version), and I did not have pyserial for python 3 installed.  Does this script depend on python 3?

I have installed pip3 and executed pip3 install pyserial.  However, after running the process again, I am getting the same Serial import error.  Also, while running catkin_make, I got a couple errors this time (sorry for the long code, just focus on on the error right at the 95% mark, and the very end):

`[ 91%] Building CXX object rosserial/rosserial_server/CMakeFiles/rosserial_server_serial_node.dir/src/serial_node.cpp.o
[ 95%] Building CXX object rosserial/rosserial_server/CMakeFiles/rosserial_server_udp_socket_node.dir/src/udp_socket_node.cpp.o
[ 95%] Building CXX object rosserial/rosserial_server/CMakeFiles/rosserial_server_socket_node.dir/src/socket_node.cpp.o
c++: internal compiler error: Killed (program cc1plus)
Please submit a full bug report,
with preprocessed source if appropriate.
See <file:///usr/share/doc/gcc-7/README.Bugs> for instructions.
rosserial/rosserial_server/CMakeFiles/rosserial_server_serial_node.dir/build.make:62: recipe for target 'rosserial/rosserial_server/CMakeFiles/rosserial_server_serial_node.dir/src/serial_node.cpp.o' failed
make[2]: *** [rosserial/rosserial_server/CMakeFiles/rosserial_server_serial_node.dir/src/serial_node.cpp.o] Error 4
CMakeFiles/Makefile2:3280: recipe for target 'rosserial/rosserial_server/CMakeFiles/rosserial_server_serial_node.dir/all' failed
make[1]: *** [rosserial/rosserial_server/CMakeFiles/rosserial_server_serial_node.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
In file included from /usr/include/c++/7/vector:64:0,
                 from /usr/include/boost/asio/detail/impl/service_registry.ipp:19,
                 from /usr/include/boost/asio/detail/service_registry.hpp:143,
                 from /usr/include/boost/asio/impl/io_service.hpp:19,
                 from /usr/include/boost/asio/io_service.hpp:767,
                 from /usr/include/boost/asio/basic_io_object.hpp:19,
                 from /usr/include/boost/asio/basic_socket.hpp:20,
                 from /usr/include/boost/asio/basic_datagram_socket.hpp:20,
                 from /usr/include/boost/asio.hpp:21,
                 from /home/ubuntu/catkin_ws/src/rosserial/rosserial_server/src/socket_node.cpp:34:
/usr/include/c++/7/bits/stl_vector.h: In member function ‘void std::vector<_Tp, _Alloc>::push_back(const value_type&) [with _Tp = boost::asio::detail::timer_queue<boost::asio::detail::forwarding_posix_time_traits>::heap_entry; _Alloc = std::allocator<boost::asio::detail::timer_queue<boost::asio::detail::forwarding_posix_time_traits>::heap_entry>]’:
/usr/include/c++/7/bits/stl_vector.h:948:21: note: parameter passing for argument of type ‘__gnu_cxx::__normal_iterator<boost::asio::detail::timer_queue<boost::asio::detail::forwarding_posix_time_traits>::heap_entry*, std::vector<boost::asio::detail::timer_queue<boost::asio::detail::forwarding_posix_time_traits>::heap_entry, std::allocator<boost::asio::detail::timer_queue<boost::asio::detail::forwarding_posix_time_traits>::heap_entry> > >’ changed in GCC 7.1
    _M_realloc_insert(end(), __x);
    ~~~~~~~~~~~~~~~~~^~~~~~~~~~~~
In file included from /usr/include/c++/7/vector:64:0,
                 from /usr/include/boost/asio/detail/impl/service_registry.ipp:19,
                 from /usr/include/boost/asio/detail/service_registry.hpp:143,
                 from /usr/include/boost/asio/impl/io_service.hpp:19,
                 from /usr/include/boost/asio/io_service.hpp:767,
                 from /usr/include/boost/asio/basic_io_object.hpp:19,
                 from /usr/include/boost/asio/basic_socket.hpp:20,
                 from /usr/include/boost/asio/basic_datagram_socket.hpp:20,
                 from /usr/include/boost/asio.hpp:21,
                 from /home/ubuntu/catkin_ws/src/rosserial/rosserial_server/src/udp_socket_node.cpp:34:
/usr/include/c++/7/bits/stl_vector.h: In member function ‘void std::vector<_Tp, _Alloc>::push_back(const value_type&) [with _Tp = boost::asio::detail::timer_queue<boost::asio::detail::forwarding_posix_time_traits>::heap_entry; _Alloc = std::allocator<boost::asio::detail::timer_queue<boost::asio::detail::forwarding_posix_time_traits>::heap_entry>]’:
/usr/include/c++/7/bits/stl_vector.h:948:21: note: parameter passing for argument of type ‘__gnu_cxx::__normal_iterator<boost::asio::detail::timer_queue<boost::asio::detail::forwarding_posix_time_traits>::heap_entry*, std::vector<boost::asio::detail::timer_queue<boost::asio::detail::forwarding_posix_time_traits>::heap_entry, std::allocator<boost::asio::detail::timer_queue<boost::asio::detail::forwarding_posix_time_traits>::heap_entry> > >’ changed in GCC 7.1
    _M_realloc_insert(end(), __x);
    ~~~~~~~~~~~~~~~~~^~~~~~~~~~~~
In file included from /usr/include/c++/7/vector:69:0,
                 from /usr/include/boost/asio/detail/impl/service_registry.ipp:19,
                 from /usr/include/boost/asio/detail/service_registry.hpp:143,
                 from /usr/include/boost/asio/impl/io_service.hpp:19,
                 from /usr/include/boost/asio/io_service.hpp:767,
[... cut ...]
/usr/include/c++/7/bits/stl_uninitialized.h:76:9: note: parameter passing for argument of type ‘std::move_iterator<boost::asio::detail::timer_queue<boost::asio::detail::forwarding_posix_time_traits>::heap_entry*>’ changed in GCC 7.1
[ 98%] Linking CXX executable /home/ubuntu/catkin_ws/devel/lib/rosserial_server/socket_node
[ 98%] Linking CXX executable /home/ubuntu/catkin_ws/devel/lib/rosserial_server/udp_socket_node
[ 98%] Built target rosserial_server_udp_socket_node
[ 98%] Built target rosserial_server_socket_node
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j4 -l4" failed
`

welkerCode, 2018-08-02:
I just tried to the following from the python console:

from serial import Serial

and it throws the same error.  However

import serial 

does not...  But, if I try to use serial.Serial, then it says 'module' object has no attribute 'Serial'

Is there a different serial module other than pyserial that I should be using?

welkerCode, 2018-08-02:
This was the page that helped me to figure it out:
https://raspberrypi.stackexchange.com/questions/74742/python-serial-serial-module-not-found-error

doisyg, 2018-08-28:
Okay, so would it be possible to release it for Melodic now that it seems to build and work when compiled from source? @mikeferguson

mikeferguson, 2018-08-28:
@doisyg -- I've not been a maintainer of rosserial for about 4 years now.

doisyg, 2018-08-31:
@mikeferguson , sorry I just read the author line of the package.xml, not the maintener one.
@PaulBouchier , any update?

4ndr3w, 2018-10-05:
Are there any plans/blockers for releasing on Melodic?

asaba96, 2018-10-08:
I would love Melodic support please. Having to manage it on our own and build it from source is a pain.

mikepurvis, 2018-10-11:
I had been hoping to merge a few more pending PRs, but this has gone on long enough; thanks for your patience everybody.

Release 0.8.0 is tagged: e9376888d2cc9faf12b3aab6a532165109431d58

Blooming in progress.

mikepurvis, 2018-10-17:
We're built: http://repositories.ros.org/status_page/ros_melodic_default.html?q=rosserial

closed 2018-10-17 by mikepurvis