ros-drivers/rosserial #390: Multiple launch of rosserial_python

nnn112358, 2018-09-09:
I tried wifi communication between one Ubuntu(16.04) and multiple ESP32 by using ROS.
Executed with the following command

Ubuntu
 >roscore
 >rosrun rosserial_python serial_node.py tcp

ESP32
 >I used the following
 >https://qiita.com/nnn112358/items/1cd4517ea2faa57e9c13#arduino%E3%81%AE%E5%AE%9F%E8%A3%85

As I confirmed,
rosserial_python seems to be doing one-on-one communication between PC and ESP.

I want to launch more than one rosserial_python, but I do not know how to specify a specific port.

Please give me some information if you have been challenged elsewhere.

romainreignier, 2018-09-09:
From the source code, you have to specify one more argument to the command line on the `rosserial_python` side:

https://github.com/ros-drivers/rosserial/blob/6b996aae81df509614c08b3c9ed9ce7a27d2c7a0/rosserial_python/nodes/serial_node.py#L67

For the ESP side, you have to call the `setConnection` method with two arguments to specify the port before the `initNode` method, like in the example:

https://github.com/ros-drivers/rosserial/blob/6b996aae81df509614c08b3c9ed9ce7a27d2c7a0/rosserial_arduino/src/ros_lib/examples/Esp8266HelloWorld/Esp8266HelloWorld.ino#L50

Saums, 2019-08-05:
Does assigning different ports to different Arduinos, allow you to have multiple connections?

romainreignier, 2019-08-16:
Yes, it should works.
Be aware of conflicts on topics names.

closed 2020-08-17 by nnn112358