ros-drivers/rosserial #398: Unable to sync with device; possible link problem or link software version mismatch such as hydro rosserial_python with groovy Arduino

chlakshminarayana, 2018-10-20:
Unable to sync with device; possible link problem or link software version mismatch such as hydro rosserial_python with groovy Arduino

I am getting this error while rosserial reading from melodic

skohlbr, 2018-11-07:
ETHZ's fork of rosserial features a [proper-time-sync](https://github.com/ethz-asl/rosserial/tree/feature/proper-time-sync) branch. I haven't tested it, but it might be something worth looking at.

nicolasarl, 2019-04-16:
I have the same problem. When I rote my code in Arduino I used kinetic version, but I updated all the system, including the ros_lib folder to melodic. Since that moment I got the error that you mentioned. 

Any suggestion?

chlakshminarayana, 2019-04-17:
check your arduino code ....
I am pretty sure ,  some blocks of your arduino code is not suporting by 'ROS' serial package .
change that lines according to ROS

senceryazici, 2019-07-12:
If the versions match, might be due to a runtime error on your hardware that prevents the device to talk to the server. Check if subscribers and publishers are set correctly, and node initializes properly, and loop is handled properly (nh.initNode() for init and nh.spinOnce() for loop), but I highly recommend you to check the ros_lib, the library that was generated for arduino, it should be generated by it's package with the correct version of ROS.

Check this link out,
http://wiki.ros.org/rosserial_arduino/Tutorials

nehagarg, 2020-09-01:
Does your rosmsg type has an array field inside it? Melodic gives this error with array msg type. See here : https://nehainblunderland.blogspot.com/2020/08/running-arduino-board-with-ubuntu-18.html

Abhinavgandhi09, 2020-11-07:
This issue might occur if you are trying to use Float32MultiArray msgs. There is some issue with it's implementation(maybe something with the buffer size) in ros_lib which causes a runtime error and thus rosserial is not able to connect to the arduino. I tried using Float64MultiArray instead and it worked.

DeepakKumarShah, 2021-09-17:
https://github.com/ros-drivers/rosserial/issues/354    try this if you are using esp, may be it will solve the problem.

javidaslam, 2022-08-10:
SOLVED : Hey, I encountered the same error , it related to baud rate ! i changed baud rate to 57600 and it worked ,error solved !

rosrun rosserial_arduino serial_node.py /dev/ttyACM0 _baud:=57600

jones2126, 2022-11-16:
@javidaslam Thank you.  baud:=57600 fixed it for me also.

limn3, 2023-03-08:
> /dev/ttyACM0 _baud:=57600

Where can this file be found?

StohanzlMart, 2023-06-10:
Had the same error (really !helpful error message).

After "rosrun rosserial_arduino make_libraries.py .":

For me the fix was to change the line in my (Platformio) project in lib/ros_lib/ros.h where just because ESP8266 hw it assumes you want to use WiFi. So the error message should have included the hint "(ros)Serial interface may be misconfigured, please check ros.h in your project." -- Someone with more patience might do a pull request on this.

Who ever had the bright idea to put this (in ros.h) there, !thank you for wasting about 10 hours of my life!

Hdriwn, 2023-06-20:
If none of the solutions mentioned above resolve the problem, consider rechecking or removing any loops from your Arduino code. I encountered a similar issue, and removing a while loop which i used in the void loop in my code resolved it. My issue was the program was getting stuck in that while loop, maybe due to something on my side regarding the exit condition, but i got the same error message.

tripa2020, 2025-05-12:
TD;LR 
Solved for **Arduino DUE, Ubuntu 20.04, Ros Noetic** after adding these lines before initializing node (nh.initNode) in the arduino setup function
  
Serial.begin(57600);
  nh.getHardware()->setPort(&Serial);

### Rosserial Problem Solving Attempts 
**Physical Connections:**
 Testing multiple (three) different USB cables, including those rated for high speed.
    - Trying different physical USB ports on my computer.
    - Using two different Arduino DUE boards to rule out a single faulty board.
    - Attempting connection via both the Arduino DUE's Programming Port and its Native USB Port (and ensuring the sketch used Serial vs. SerialUSB accordingly).
    - Trying an external TTL-to-USB converter.

**Software & Configuration :**
    - Verifying basic serial communication stability using the Arduino IDE's Serial Monitor 
    - Updating all ROS Noetic packages using sudo apt update and sudo apt upgrade 
    - Subsequently, completely deleting the ros_lib folder from my Arduino sketchbook and regenerating it using rosrun rosserial_arduino make_libraries.py . (ensuring my ROS workspace was correctly sourced).
    - Experimenting with different baud rates (e.g., 9600, 115200) in both the Arduino sketch and the serial_node.py command.
    - Attempting to alter SerialClient.py within the rosserial_python package to get more insight into protocol byte exchanges during the handshake.

I was so annoyed just trying anything, and randomly I tried this function that auto completed in  the Arduino IDE and suddenly it worked. After investigating why it worked it seems to me that the issue stems from ArduinoHardware.h and some ambiguity in the initialization of the Serial port for the Arduino. By explicitly setting the internal iostream pointer to the Serial object that was just initialized it seems to have solved all the issues. 

Hopes this helps a frustrated engineer!