ros-drivers/rosserial #406: Can't start UDP socket: Buffer overrun when attempting to parse setup message.

fdziwuls, 2018-12-05:
I get the following when attempting to launch a udp_socket:

[ERROR] [1543967945.196795346]: Buffer overrun when attempting to parse setup message.
[ERROR] [1543967945.196814178]: Is this firmware from a pre-Groovy rosserial?

Running on Ubuntu 16.04.

jessicaaustin, 2021-11-20:
Same issue with melodic and noetic. 

```
$ docker build -t rosserial-melodic -f Dockerfile.melodic .
$ docker run --rm --net host rosserial-melodic 
[ERROR] [1637369060.168613181]: Buffer overrun when attempting to parse setup message.
[ERROR] [1637369060.168632815]: Is this firmware from a pre-Groovy rosserial?
[ERROR] [1637369062.168936526]: Buffer overrun when attempting to parse setup message.
[ERROR] [1637369064.169109883]: Buffer overrun when attempting to parse setup message.
[ERROR] [1637369066.169416465]: Buffer overrun when attempting to parse setup message.

```

```
# Dockerfile.melodic
FROM ros:melodic-ros-base-bionic

RUN apt-get update \
    && apt-get install -y ros-melodic-rosserial ros-melodic-rosserial-client ros-melodic-rosserial-msgs ros-melodic-rosserial-python ros-melodic-rosserial-server  \
    && rm -rf /var/lib/apt/lists/*

CMD roslaunch rosserial_server udp_socket.launch
```

```
# Dockerfile.noetic
FROM ros:noetic-ros-base-focal

RUN apt-get update \
    && apt-get install -y ros-noetic-rosserial ros-noetic-rosserial-client ros-noetic-rosserial-msgs ros-noetic-rosserial-python ros-noetic-rosserial-server  \
    && rm -rf /var/lib/apt/lists/*

CMD roslaunch rosserial_server udp_socket.launch
```

SaleJuice, 2023-03-15:
Same issue to me. 
Platform: ESP32C3 + Arduino + ROS noetic + Ubuntu20.04