ros-drivers/rosserial #408: rosserial_python : Service handler returned None

michalrom089, 2018-12-18:
I've setup service server on stm32 :

`
ros::NodeHandle nh;

void srv_callback(const custom_srv::AddTwoIntsRequest &req, custom_srv::AddTwoIntsResponse &res){
	res.Sum = 2;
}

ros::ServiceServer<custom_srv::AddTwoIntsRequest,custom_srv::AddTwoIntsResponse> server("add_two_ints",&srv_callback);

void setup(){
	nh.initNode();
	nh.advertiseService(server);
}
void loop(void)
{
	nh.spinOnce();
}

`

But when I try to call the service from rosserial_python error occurs :
 `$rosservice call /add_two_ints "A: 0
B: 0" 
ERROR: service [/add_two_ints] responded with an error: service cannot process request: service handler returned None
`

I tried to debug this error on stm32. Apparently, the error occurs befor computing srv_callback(). I assume that this is related with rosserial_python package but it's onyl guessing.

jimmyshe, 2018-12-20:
I am facing the same issue, using arduino due and 8.0.0 version of rosserial.  
``` c
/*
 * rosserial Service Server
 */

#include <ros_m.h>
#include <std_msgs/String.h>
#include <rosserial_arduino/Test.h>

ros::NodeHandle  nh;
using rosserial_arduino::Test;

int i;
void callback(const Test::Request & req, Test::Response & res){
  if((i++)%2)
    res.output = "hello";
  else
    res.output = "world";
}

ros::ServiceServer<Test::Request, Test::Response> server("test_srv",&callback);

std_msgs::String str_msg;
ros::Publisher chatter("chatter", &str_msg);

char hello[13] = "hello world!";

void setup()
{
  nh.initNode();
  nh.advertiseService(server);
  nh.advertise(chatter);
}

void loop()
{
  str_msg.data = hello;
  chatter.publish( &str_msg );
  nh.spinOnce();
  delay(10);
}
``` 

``` bash
$rosservice call /test_srv "input: 'sssss'"
ERROR: service [/test_srv] responded with an error: service cannot process request: service handler returned None

betab0t, 2019-01-20:
Hi all,

Did some tests with my Arduino Mega 2560... 0.7.7 works fine while 0.8.0 returns `service handler returned None`...

After trying different versions in between 0.7.7-0.8.0 I can tell that this issue was introduced in https://github.com/ros-drivers/rosserial/commit/fe1e233f39d56931b5e223c5e53a2d9c2f9fb664.

Tones29, 2019-01-24:
I got the same issue with an Arduino Micro: 0.7.7 is working, 0.8.0 responds with the error.

doisyg, 2019-02-07:
@chrisspen and @mikepurvis 
Since @betab0t seems to have identified the bug from the PR https://github.com/ros-drivers/rosserial/commit/fe1e233f39d56931b5e223c5e53a2d9c2f9fb664 , do you see a way to fix it?

lenzc, 2019-02-26:
I proposed a fix in #414.

camisatx, 2020-05-27:
Any updates on getting #414 merged, or suggestions for downgrading to 0.7.7? I can't find 0.7.7 on `apt` so I'm having trouble with the easy downgrade route.