ros-drivers/rosserial #409: Add version number for the library

nemo8, 2018-12-28:
Currently rosserial lib does not have version number in library files. So it is very difficult no know which version of lib is used. 

My propousal to add version of library to the header for each files like CMSIS or FreeRTOS has:

"@version  V4.30"
" FreeRTOS V9.0.0 - Copyright (C) 2016 Real Time Engineers Ltd."
[image]

[image]

nemo8, 2019-01-03:
I think a file version can be taken from "rosserial_client/package.xml" file - the tag "version>0.8.0</version>".

Could someone ,please, explain me how the header information is generated for each rosserial lib files? Is there any template file for this purpose?

romainreignier, 2019-01-04:
> Could someone ,please, explain me how the header information is generated for each rosserial lib files? Is there any template file for this purpose?

For the core `rosserial` files, they are not generated but the files are here: https://github.com/ros-drivers/rosserial/tree/melodic-devel/rosserial_client/src/ros_lib

For the header files generated for each message, they are generated from this Python script: 
https://github.com/ros-drivers/rosserial/blob/611ed413de06f1218d09317594eecfc39655d7ed/rosserial_client/src/rosserial_client/make_library.py#L410

So if you want want to add comments at the beginning of those files, you should look in that Python script.

And the core files are copied there so you also can add headers to them:
https://github.com/ros-drivers/rosserial/blob/611ed413de06f1218d09317594eecfc39655d7ed/rosserial_client/src/rosserial_client/make_library.py#L574

romainreignier, 2019-02-24:
I could not find any API to retrieve the version of a given package but `rosversion` tool is able to do it, see here:

https://github.com/ros-infrastructure/rospkg/blob/e93f276ec636839d887e0be775db40fe80372c62/src/rospkg/rosversion.py#L105-126