ros-drivers/rosserial #411: rosserial_server: Never establishes connection

raabuchanan, 2019-02-01:
When I run the following command in terminal: `rosrun rosserial_server serial_node _port:=/dev/ttyACM0 _baud:=57600`

I get repeated messages:
`Opened /dev/ttyACM0
Opened /dev/ttyACM0
Opened /dev/ttyACM0
Opened /dev/ttyACM0`

But I never see the topics the Arduino is supposed to be advertising. It's as if it can't establish a connection. I am also running message_info_service.py. It works if I use rosserial_python with the exact same parameters but I want to switch to the supposedly more stable rosserial_server.

Ubuntu 16.04 + Kinetic + Arduino Uno

doisyg, 2019-02-12:
Same issue with 18.04 + Melodic + Arduino Uno

andre-nguyen, 2019-03-11:
Same Ubuntu 18.04.2 LTS + STM32L4

Looks like the checksum never passes. Have you guys investigated further?

tuandle, 2019-05-07:
I have the same problem with 16.04 + Kinetic + Arduino Mega 2560. 
Have you guys found any workaround? 
I already tried to build from source but the problem is the same.

andre-nguyen, 2019-05-07:
Never managed to figure it out and ended up using the Python version. Still a mystery to me.

raabuchanan, 2019-05-08:
Like @andre-nguyen I gave up trying to find a fix and used the Python version.

hamma95, 2019-05-28:
same issue here with 18.04 ros melodic and arduino uno. could it be the cables?

romainreignier, 2019-05-31:
Usaually, when I have the issue, either there is an issue with the connection cables or the firmware has crashed and do not call `nh.spinOnce()` fast enough.

hamma95, 2019-06-03:
the same code works fine with rosserial python though, and with the socket node of rosserial server. as for the cables the cable works with rosserial python too. i really want to switch to rosserial server tho because its (supposedly) more robust and less prone to errors

jsseng, 2020-03-23:
I have the same problem as well.  Did anyone ever figure it out?

Jenifen, 2020-09-22:
try to launch the node server  twice :
rosrun rosserial_server serial_node _port:=/dev/ttyACM0 &
rosrun rosserial_server serial_node _port:=/dev/ttyACM0 

works for me !