ros-drivers/rosserial #423: I can't receive message when the message typeis sensor_msgs::Image ,

cenjinchang, 2019-04-28:
I write a ros::Subcriber (the code bellow)，the topic type is sensor_msgs::Image ,receive nothing ,but when I change the type to std_msgs::String ,everything is ok .

and ,I am sure that the ros::Publisher is ok in ubuntu 16.04  that I can see the message detial when use command "rostopic echo /say_hello_world".

here is the Subcriber code （use rosserial_windows）
void get_image_callback (const sensor_msgs::Image & image)
{
  printf ("Received : height(%d),width(%d),encoding(%s)\n", image.height,image.width,image.encoding);
}

int main (int argc, char * argv[])
{
  ros::NodeHandle nh;
  char *ros_master = "*****";

  printf ("Connecting to server at %s\n", ros_master);
  nh.initNode (ros_master);
  ros::Subscriber <sensor_msgs::Image>
          imageSub("say_hello_world", &get_image_callback);
  nh.subscribe(imageSub);

  printf ("Waiting to receive messages\n");
  while (1)
  {
    nh.spinOnce ();
    Sleep (1);
  }
  printf ("All done!\n");
  return 0;
}

and the Publisher code （in Ubuntu）
int main(int argc, char** argv)
{
  ros::init(argc, argv, "image_publisher");
  ros::NodeHandle nh;
  cv::Mat image = cv::imread("/home/cjc/catkin_ws/src/image_transport_pkg/src/image.png", CV_LOAD_IMAGE_GRAYSCALE);
 
  sensor_msgs::ImagePtr msg = cv_bridge::CvImage(std_msgs::Header(), "mono8", image).toImageMsg();
  sensor_msgs::Image imageMsg = *msg;
  image_transport::ImageTransport it(nh);
  image_transport::Publisher pub = it.advertise("say_hello_world", 1);
  while (ros::ok()) {
    pub.publish(imageMsg);
    ros::spinOnce();
    sleep(10);
  }
}

cenjinchang, 2019-04-28:
The default template arguments of ros::NodeHandle configure an input buffer of 512 bytes

That is why the question appear. but how to modify the buffer ?

romainreignier, 2019-04-30:
Indeed, you have to modify the template arguments directly in the lib.
But be aware that the message must have a size smaller than 65 kB, see my comment [here](https://github.com/ros-drivers/rosserial/issues/341#issuecomment-350059896).