ros-drivers/rosserial #427: Error - no attribute 'flushoutput'

remintz-acn, 2019-06-19:
When running 

`$ rosrun rosserial_python serial_node.py tcp`

and the device (NodeMCU) is reset

I get the following 

```
[ERROR] [1560968878.473622]: Lost sync with device, restarting...
[INFO] [1560968878.475121]: Requesting topics...
[WARN] [1560968878.498287]: Last read step: data checksum
[WARN] [1560968878.499542]: Run loop error: [Errno 104] Connection reset by peer
[INFO] [1560968878.510628]: Removing subscriber: /hero_0/cmd_vel
[INFO] [1560968878.512802]: Removing subscriber: /hero_0/motor
[INFO] [1560968878.514852]: Shutting down
[INFO] [1560968878.515763]: All done
Traceback (most recent call last):
  File "/opt/ros/melodic/lib/rosserial_python/serial_node.py", line 73, in <module>
    server.listen()
  File "/opt/ros/melodic/lib/python2.7/dist-packages/rosserial_python/SerialClient.py", line 253, in listen
    self.startSerialClient()
  File "/opt/ros/melodic/lib/python2.7/dist-packages/rosserial_python/SerialClient.py", line 259, in startSerialClient
    client.run()
  File "/opt/ros/melodic/lib/python2.7/dist-packages/rosserial_python/SerialClient.py", line 556, in run
    self.port.flushOutput()
AttributeError: RosSerialServer instance has no attribute 'flushOutput'

```

Saums, 2019-09-16:
I'm receiving the same error. Did you find a solution?

Koraze, 2019-11-11:
I work on ROS Melodic on Raspberry Pi 4 with Raspbian Buster.

I also got the same issue, so I remove code around line 556 in SerialClient.py
```
with self.write_lock:
self.port.flushOutput()
```
It's seems to work now ... I hope those lines weren't so important ...

mbilsky, 2020-04-29:
On line 281 flushInput() is defined, so I pasted a similar definition right below:

    def flushInput(self):
        pass

    def flushOutput(self):
        pass

I initially thought it may be because flushInput/flushOutput are depreciated from pyserial (now called reset_output_buffer() ): https://pythonhosted.org/pyserial/pyserial_api.html#serial.Serial.reset_output_buffer

I'm running pyserial 3.4. This change didn't help so I just created the blank definition above and it seems to work robustly enough.