ros-drivers/rosserial #429: only works with Twist, not TwistStamped

nathangeorge1, 2019-06-27:
In testing rosserial_python/serial_node.py, it appears only Twist messages successfully activate the callback, not TwistStamped. 

See another related problem: https://answers.ros.org/question/250366/arduino-rosserial-node-hangs-during-i2c-reading-pwm/?answer=327171#post-id-327171

Thanks,
Nathan

romainreignier, 2019-07-01:
Could share you sketch with us?

Note that a subscriber or publisher only works for one type of message so either `Twist` or `TwistStamped` but not both.

nathangeorge1, 2019-07-02:
Steps to reproduce:
1. upload the below code
2. rosrun rosserial_python serial_node.py _port:=/dev/ttyACM0 _baud:=57600

It doesn't subscribe to the TwistStamped one; here is the output:
```
[N=/serial_node,F=<module>-49]: ROS Serial Python Node
[N=/serial_node,F=<module>-86]: Connecting to /dev/ttyACM0 at 57600 baud
[N=/serial_node,F=SerialClient.requestTopics-404]: Requesting topics...
[N=/serial_node,F=SerialClient.setPublishSize-562]: Note: publish buffer size is 512 bytes
[N=/serial_node,F=SerialClient.setupPublisher-578]: Setup publisher on /motor_controller/messages [std_msgs/String]
[N=/serial_node,F=SerialClient.setSubscribeSize-567]: Note: subscribe buffer size is 512 bytes
[N=/serial_node,F=SerialClient.setupSubscriber-591]: Setup subscriber on /motor_controller/command_no_stamp [geometry_msgs/Twist]
^C[N=/serial_node,F=SerialClient.txStopRequest-423]: Send tx stop request
```

```
// Arduino Includes
#include <Arduino.h>              // should be first
#include <Wire.h>

// ROS Includes
#include <ros.h>  
#include <geometry_msgs/Twist.h>
#include <geometry_msgs/TwistStamped.h> // doesn't work
#include <std_msgs/String.h>

// Namespaces
using namespace std;

// Globals   
std_msgs::String message;
ros::Publisher pub_messages("/motor_controller/messages", &message);
ros::NodeHandle nh;

// Callbacks
void commandCallback(const geometry_msgs::Twist& msg) { 
    message.data = "Twist message received";
    pub_messages.publish(&message);   
}
ros::Subscriber<geometry_msgs::Twist> sub_command("/motor_controller/command_no_stamp", &commandCallback, 1);

void command2Callback(const geometry_msgs::TwistStamped& msg) { 
    message.data = "TwistStamped message received";
    pub_messages.publish(&message);   
}
ros::Subscriber<geometry_msgs::TwistStamped> sub_command2("/motor_controller/command", &command2Callback, 1);

void setup() {
    // general init
    nh.initNode();
    nh.subscribe(sub_command); 
    nh.advertise(pub_messages);
    Serial.begin(57600);
}

void loop() { 
  
    delay(1);
    nh.spinOnce();

}
```

romainreignier, 2019-07-03:
The following line is missing in your code:

```C++
nh.subscribe(sub_command2);
```

nathangeorge1, 2019-07-05:
ah yes, forgot that; but even with it, publishing to sub_command activates the callback, but not for sub_command2:

serial_node.py (note TwistStamped publisher started):
```
rosrun rosserial_python serial_node.py _port:=/dev/ttyACM1
[N=/serial_node,F=<module>-49]: ROS Serial Python Node
[N=/serial_node,F=<module>-86]: Connecting to /dev/ttyACM1 at 57600 baud
[N=/serial_node,F=SerialClient.requestTopics-404]: Requesting topics...
[N=/serial_node,F=SerialClient.setPublishSize-562]: Note: publish buffer size is 512 bytes
[N=/serial_node,F=SerialClient.setupPublisher-578]: Setup publisher on /motor_controller/messages [std_msgs/String]
[N=/serial_node,F=SerialClient.setSubscribeSize-567]: Note: subscribe buffer size is 512 bytes
[N=/serial_node,F=SerialClient.setupSubscriber-591]: Setup subscriber on /motor_controller/command_no_stamp [geometry_msgs/Twist]
[N=/serial_node,F=SerialClient.setupSubscriber-591]: Setup subscriber on /motor_controller/command [geometry_msgs/TwistStamped]
^C[N=/serial_node,F=SerialClient.txStopRequest-423]: Send tx stop request
```

`rostopic echo /motor_controller/messages` (I published to both subscribers)
```
^Cnathan@nathan-vm:~$ rostopic echo /motor_controller/messages 
data: "Twist message received"
---
^Cnathan@nathan-vm:~$ 
```

full code:
```
// Arduino Includes
#include <Arduino.h>              // should be first
#include <Wire.h>

// ROS Includes
#include <ros.h>  
#include <geometry_msgs/Twist.h>
#include <geometry_msgs/TwistStamped.h> // doesn't work
#include <std_msgs/String.h>

// Namespaces
using namespace std;

// Globals   
std_msgs::String message;
ros::Publisher pub_messages("/motor_controller/messages", &message);
ros::NodeHandle nh;

// Callbacks
void commandCallback(const geometry_msgs::Twist& msg) { 
    message.data = "Twist message received";
    pub_messages.publish(&message);   
}
ros::Subscriber<geometry_msgs::Twist> sub_command("/motor_controller/command_no_stamp", &commandCallback, 1);

void command2Callback(const geometry_msgs::TwistStamped& msg) { 
    message.data = "TwistStamped message received";
    pub_messages.publish(&message);   
}
ros::Subscriber<geometry_msgs::TwistStamped> sub_command2("/motor_controller/command", &command2Callback, 1);

void setup() {
    // general init
    nh.initNode();
    nh.subscribe(sub_command); 
    nh.subscribe(sub_command2);
    nh.advertise(pub_messages);
    Serial.begin(57600);
}

void loop() { 
  
    delay(1000);
    nh.spinOnce(); 

}
```

romainreignier, 2019-07-07:
Hum... That's very strange.
On which board did you tried?
I will try to reproduce the issue.

nathangeorge1, 2019-07-07:
Arduino Mega 2560

romainreignier, 2019-07-10:
I have finally found the time to test your code on an Arduino Mega 2560 and found the same issue.
But reducing the delay in the loop to 10 ms made it work.
In general, `spinOnce()` should be called very regularly.
But I am not sure yet why increasing this delay make the communication fail. Even at 20 ms the answer does not arrive.

nathangeorge1, 2019-07-11:
I also reproduced your findings. 

My robot system needs to calls spinOnce at 150ms intervals due to high (40-100ms) latency between the robot and the main computer.

My guess is that the Twist being only 6 floats is easier to pass into the Arduino than the Twist plus the Header, which has the string and time types. I also know [ROS arrays](https://docs.ros.org/api/std_msgs/html/msg/Int32MultiArray.html) cannot be passed to Arduino; so it could be a bandwidth or checksum issue.

Do you have any ideas of a workaround to use TwistStamped? I thought of sending the Twist and Header messages separately.