ros-drivers/rosserial #431: Arduino Nano Every Support

k-kleber, 2019-07-02:
Arduino just released its new Nano variant EVERY 
[Nano Every](https://store.arduino.cc/nano-every)
It uses the ATMega4809 processor, but is not supported using rosserial-arduino yet.

If someone could add the compile files for it it would be amazing, as I could not figure out how to do it myself.

Thanks

romainreignier, 2019-07-02:
Hi @k-kleber
Did you actually tried current version of `rosserial_arduino` on it?
Because thanks of the Arduino abstraction API, it should work out of the box.

If it does not work, maybe the `HardwareSerial` type does not exist.
In that case, this part should be modified if include a specific section:

https://github.com/ros-drivers/rosserial/blob/ae3366efafce82ec36fd730c604fd0b55d7bff0e/rosserial_arduino/src/ros_lib/ArduinoHardware.h#L54,L63

k-kleber, 2019-07-03:
Hey @romainreignier 
I managed to compile for the board by using some small changes to arduino-mk and using that directly. There is a pull request for that in the Aruino-Makefile reposistory for this.
I do, however, have a problem with the ros_lib where I am hoping that someone might know the solution.
I created a very basic test program only creating a node_handle where it complains about:
` undefined reference to 'Print::write(unsigned char const*, unsigned int)' `

Source Code:
```
#include <Arduino.h>
#include <ros.h>

void setup() {
  ros::NodeHandle nh;
}

void loop() {}
```

Makefile
```
ARDUINO_DIR = $(HOME)/arduino-1.8.9
BOARD_TAG = nona4809
ALTERNATE_CORE_PATH = $(HOME)/.arduino15/packages/arduino/hardware/megaavr/1.8.2
ARDUINO_LIBS = ros_lib
include /home/kevin/Arduino-Makefile/Arduino.mk
```

Make Output (Make ran 2 times to remove non-error related ouput)
```
-------------------------
Arduino.mk Configuration:
- [AUTODETECTED]       CURRENT_OS = LINUX 
- [USER]               ARDUINO_DIR = /home/kevin/arduino-1.8.9 
- [COMPUTED]           ARDMK_DIR = /home/kevin/Arduino-Makefile (relative to Common.mk)
- [AUTODETECTED]       ARDUINO_VERSION = 189 
- [DEFAULT]            ARCHITECTURE = avr 
- [DEFAULT]            ARDMK_VENDOR = arduino 
- [AUTODETECTED]       ARDUINO_PREFERENCES_PATH = /home/kevin/.arduino15/preferences.txt 
- [AUTODETECTED]       ARDUINO_SKETCHBOOK = /home/kevin/Arduino (from arduino preferences file)
- [USER]               ALTERNATE_CORE_PATH = /home/kevin/.arduino15/packages/arduino/hardware/megaavr/1.8.2 
- [COMPUTED]           ARDUINO_VAR_PATH = /home/kevin/.arduino15/packages/arduino/hardware/megaavr/1.8.2/variants (from ALTERNATE_CORE_PATH)
- [COMPUTED]           BOARDS_TXT = /home/kevin/.arduino15/packages/arduino/hardware/megaavr/1.8.2/boards.txt (from ALTERNATE_CORE_PATH)
- [BUNDLED]            AVR_TOOLS_DIR = /home/kevin/arduino-1.8.9/hardware/tools/avr (in Arduino distribution)
- [COMPUTED]           ARDUINO_LIB_PATH = /home/kevin/arduino-1.8.9/libraries (from ARDUINO_DIR)
- [COMPUTED]           ARDUINO_PLATFORM_LIB_PATH = /home/kevin/arduino-1.8.9/hardware/arduino/avr/libraries (from ARDUINO_DIR)
- [DEFAULT]            USER_LIB_PATH = /home/kevin/Arduino/libraries (in user sketchbook)
- [DEFAULT]            PRE_BUILD_HOOK = pre-build-hook.sh 
- [USER]               BOARD_TAG = nona4809 
- [COMPUTED]           CORE = arduino (from build.core)
- [COMPUTED]           VARIANT = nona4809 (from build.variant)
- [COMPUTED]           OBJDIR = build-nona4809 (from BOARD_TAG)
- [COMPUTED]           ARDUINO_CORE_PATH = /home/kevin/.arduino15/packages/arduino/hardware/megaavr/1.8.2/cores/arduino (from ALTERNATE_CORE_PATH, BOARD_TAG and boards.txt)
- [ASSUMED]            MONITOR_BAUDRATE = 9600 
- [DEFAULT]            OPTIMIZATION_LEVEL = s 
- [DEFAULT]            MCU_FLAG_NAME = mmcu 
- [DEFAULT]            CFLAGS_STD = -std=gnu11 
- [DEFAULT]            CXXFLAGS_STD = -std=gnu++11 
- [AUTODETECTED]       DEVICE_PATH =  
- [DEFAULT]            FORCE_MONITOR_PORT =  
- [AUTODETECTED]       Size utility: AVR-aware for enhanced output
-
-                      ARDUINO_LIBS =
- [USER]                 ros_lib
- [COMPUTED]           BOOTLOADER_PARENT = /home/kevin/arduino-1.8.9/hardware/arduino/avr/bootloaders (from ARDUINO_DIR)
- [COMPUTED]           ARDMK_VERSION = 1.6 
- [COMPUTED]           CC_VERSION = 5.4.0 (avr-gcc)
-------------------------
mkdir -p build-nona4809
/home/kevin/arduino-1.8.9/hardware/tools/avr/bin/avr-g++ -x c++ -include Arduino.h -MMD -c -D__PROG_TYPES_COMPAT__ -mmcu=atmega4809 -DF_CPU=16000000L -DARDUINO=189 -DARDUINO_ARCH_AVR -I/home/kevin/.arduino15/packages/arduino/hardware/megaavr/1.8.2/cores/arduino -I/home/kevin/.arduino15/packages/arduino/hardware/megaavr/1.8.2/cores/arduino/api -I/home/kevin/.arduino15/packages/arduino/hardware/megaavr/1.8.2/variants/nona4809     -I/home/kevin/Arduino/libraries/ros_lib -Wall -ffunction-sections -fdata-sections -Os -fpermissive -fno-exceptions -std=gnu++11 -fno-threadsafe-statics -flto -fno-devirtualize -fdiagnostics-color=always tester.ino -o build-nona4809/tester.ino.o
/home/kevin/arduino-1.8.9/hardware/tools/avr/bin/avr-gcc -mmcu=atmega4809 -Wl,--gc-sections -Os -flto -fuse-linker-plugin -o build-nona4809/tester.elf build-nona4809/tester.ino.o   build-nona4809/libcore.a -lc -lm 
/tmp/ccsWxgLi.ltrans0.ltrans.o:(.rodata+0x46): undefined reference to `Print::write(unsigned char const*, unsigned int)'
collect2: error: ld returned 1 exit status
/home/kevin/Arduino-Makefile/Arduino.mk:1669: recipe for target 'build-nona4809/tester.elf' failed
make: *** [build-nona4809/tester.elf] Error 1
```

gismo07, 2020-06-09:
@k-kleber can you please explain, how you compile for the every board? We have the same problem right now, that rosserial is not working for the every.

realbigmitchell, 2020-06-15:
It looks like this is the pull request @k-kleber used to fix the issue:

https://github.com/sudar/Arduino-Makefile/issues/607