ros-drivers/rosserial #440: Check ROS message for BAUDRATE

theo574, 2019-09-08:
Hi!

I'm new with ROS and I ask sorry if mine is stupid question.
I'm using rosserial for publish via serial an Int16 from Arduino sketch running on STM32 F767ZI and send, via serial, another Int16 from computer to STM32.
I need to check topic message dimension for set correct baud rate, I'm sorry on google i not found something for me, really I've not understood with precision what ROS put in one message.

romainreignier, 2019-09-11:
Hi,
I don't really understand your issue.
The rosserial protocol is defined here: https://wiki.ros.org/rosserial/Overview/Protocol
For the message data, it uses the classic ROS serialization.

theo574, 2019-09-11:
Hi!
Thank you! I haven't found this page before.
I'm again sorry, for sure, if i have to transmit only one Int16, the length of my message is 10 bytes?