ros-drivers/rosserial #455: Teensy 4.0 Support

shakeebbb, 2019-11-24:
Teensy now have 4.0 board and compiling the HelloWorld (or compiling any code that includes ros.h) is giving the error

error: cannot convert 'usb_serial_class*' to 'HardwareSerial*' in assignment
       iostream = &Serial;
                ^
Multiple libraries were found for "ros.h"

Following an older thread (https://github.com/ros-drivers/rosserial/issues/259), I fixed it by modifying the line
 #if defined(__MK20DX128__) || defined(__MK20DX256__) || defined(__MK64FX512__) || defined(__MK66FX1M0__) || defined(__MKL26Z64__)
to
#if defined(__MK20DX128__) || defined(__MK20DX256__) || defined(__MK64FX512__) || defined(__MK66FX1M0__) || defined(__MKL26Z64__) || defined(__IMXRT1062__).

Is this the right solution to the issue? Can Teensy 4.0 support be included in rosserial?

romainreignier, 2019-11-24:
Yes, this seems like the correct way to do it. You should open a Pull Request.
Thanks

shakeebbb, 2019-11-24:
Great, thanks! Pull requested

mikepurvis, 2020-04-17:
Sorry for the delay on this, but it should now be fixed in `noetic-devel` and will also be patched onto `melodic-devel` and released for Melodic in coming weeks.

closed 2020-04-17 by mikepurvis

dr-parker, 2020-08-05:
@mikepurvis Didn't want to re-open this ticket, but I was checking melodic-devel and didn't see an update? I've been installing from the package manager (running melodic), but wasn't sure if the patch had been made or released for melodic. Is there an update?

fjp, 2020-11-25:
On ROS noetic, I am trying to compile a script for the Teensy 4.0 that works on Teensy 3.2. Compiling for 4.0 gives me the same error as described above:

```
In file included from /home/fjp/Arduino/libraries/Rosserial_Arduino_Library/src/ros.h:39:0,
                 from /home/fjp/git/diffbot/diffbot_base/scripts/encoders/encoders/encoders.ino:12:
/home/fjp/Arduino/libraries/Rosserial_Arduino_Library/src/ArduinoHardware.h: In constructor 'ArduinoHardware::ArduinoHardware()':
/home/fjp/Arduino/libraries/Rosserial_Arduino_Library/src/ArduinoHardware.h:54:16: error: cannot convert 'usb_serial_class*' to 'HardwareSerial*' in assignment
       iostream = &Serial;
                ^
Error compiling for board Teensy 4.0.
``` 

I have checked for updates in the Arduino IDE but it seems that I have the latest rosserial 0.7.9 installed. Anything else that I need to update?

marcomasa, 2020-12-04:
It may seem kind of obvious, but for everyone who doesn't get the missing part in the define:

Line 44 in ArduinoHardware.h should be:

```
#if defined(__MK20DX128__) || defined(__MK20DX256__) || defined(__MK64FX512__) || defined(__MK66FX1M0__) || defined(__MKL26Z64__) || defined (__IMXRT1062__)
```

Your comment was missing the underscores bc you didn't mark the lines as code in your comment.
Thanks for the solution anyways!