ros-drivers/rosserial #458: rosserial server doesn't work when used by roslaunch

PaddyCube, 2020-01-10:
Hello,

I'm developing a robot which uses a ESP8266. ROS runs on a virtual machine on my PC. I was able to get a simple HelloWorld example running over wifi but for some reasons, it acts different depending on how I start rosserial server.

If i call it this way
`rosrun rosserial_python serial_node.py tcp`

it works as expected. Topics are registered correctly and I can echo messages.

However, if I use this command
`roslaunch rosserial_server socket.launch`

it seems fine, ESP8266 device reported that it is able to connect. But there are no topics registered and also no echo.

romainreignier, 2020-01-13:
Note that these two lines are calling two different nodes, the first one is [this Python node](https://github.com/ros-drivers/rosserial/blob/melodic-devel/rosserial_python/nodes/serial_node.py) and in the second, the launch call the [rosserial_server C++ node](https://github.com/ros-drivers/rosserial/blob/melodic-devel/rosserial_server/src/socket_node.cpp).

To keep using the Python node, you should write the launch file yourself.
It should looks like:

```xml
<launch>
  <node pkg="rosserial_python" type="serial_node.py" name="rosserial_python" args="tcp">
</launch>
```

PaddyCube, 2020-01-13:
Hello,
thanks for clarification. But I'm still confused why they act different. From my understanding of ROS (I'm totally beginner), it should not matter if it is written in C++ or Python.

romainreignier, 2020-01-13:
Indeed, the language itself does not matter but here, the implementation is different.

What I mean is that it is not so strange to see differences switching between both.

Make sure to use the correct settings (port and address). Also note that rosserial_server recently changed. If you are not using the latest version from git, make sure that the `message_info_service.py` node is also launched.

viso951, 2021-03-13:
@PaddyCube I am experiencing problems connecting my ESP to the rosserver, can you specify please the ROS, ubuntu and rosserial version you are using?

romainreignier, 2021-03-15:
I am not currently using rosserial on ESP. But last time I tried was on Ubuntu 18.04, ROS Melodic and released version of rosserial.

hariharan382, 2022-01-20:
> Hello,
>
> I'm developing a robot which uses a ESP8266. ROS runs on a virtual machine on my PC. I was able to get a simple HelloWorld example running over wifi but for some reasons, it acts different [...]

Can you please paste working code ,I am confused a lot because there is no proper documentation for this issue

Boring-Name, 2023-04-28:
I got the same issue with `roslaunch rosserial_server socket.launch`. My setup is STM32 (FreeRTOS+LwIP) -- Melodic. Issue solved with using `rosrun rosserial_python serial_node.py tcp`. (Didn't know that `rosserial_python` also has support for ETH connection.)
It seems the `rosserial_server` stops listening the port and causes LwIP `ERR_RST` and then `ERR_CONN` error codes. Same error codes can be replicated through a test TCP connection with server side shutdown when client keeps sending. Or maybe some other reason that hide behind.