ros-drivers/rosserial #459: Flash error: avrdude: stk500_recv(): programmer is not responding avrdude: stk500_getsync() attempt 1 of 10: not in sync: resp=0x00

Tigerzuo, 2020-01-14:
Hello,
I am trying to flash my Arduino Nano and I can't figure out where the error is coming from. The error is avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 1 of 10: not in sync: resp=0x00
This prevents me from flashing at all. 

Thanks for reading!

mikepurvis, 2020-01-15:
Are you able to flash this Arduino in other ways, eg from Arduino IDE, or using avrdude on the command line?

I'm assuming you're connected via the Micro USB port— are you certain you have permissions set up correctly? See: https://www.arduino.cc/en/Guide/Linux#toc6

romainreignier, 2020-01-16:
Try to press the RESET button at the beginning of the upload process.

trov4, 2021-01-12:
Hey, I've been running into this issue (and a similar one) and was hoping to see if there has been any progress on this, here's some info about what I'm doing: 

- I'm running the [rosserial-arduino cmake tutorial](http://wiki.ros.org/rosserial_arduino/Tutorials/CMake) **NOTE**: this is apart of a larger package, so my rospackage name is `arduino-ros` and not `helloworld` as indicated in the tutorial...I have made sureto update everything as necesarry
- I have been able to run and deploy this via the arduino IDE 
- Yes I tried looking through the link in [@mikepurvis' comment](https://github.com/ros-drivers/rosserial/issues/459#issuecomment-574697540) 
- I have tried deploying to an Arduino Uno follwing the instructions in that tutorial, and I get the following error when attepmpting to run `make arduino-ros_firmware_hello-upload` :  
```
Scanning dependencies of target hello-upload
avrdude: stk500_getsync() attempt 1 of 10: not in sync: resp=0x00
avrdude: stk500_getsync() attempt 2 of 10: not in sync: resp=0x00
avrdude: stk500_getsync() attempt 3 of 10: not in sync: resp=0x00
avrdude: stk500_getsync() attempt 4 of 10: not in sync: resp=0x00
avrdude: stk500_getsync() attempt 5 of 10: not in sync: resp=0x00
avrdude: stk500_getsync() attempt 6 of 10: not in sync: resp=0x00
avrdude: stk500_getsync() attempt 7 of 10: not in sync: resp=0x00
avrdude: stk500_getsync() attempt 8 of 10: not in sync: resp=0x00
avrdude: stk500_getsync() attempt 9 of 10: not in sync: resp=0x00
avrdude: stk500_getsync() attempt 10 of 10: not in sync: resp=0x00

avrdude done.  Thank you.

CMakeFiles/hello-upload.dir/build.make:57: recipe for target 'CMakeFiles/hello-upload' failed
make[7]: *** [CMakeFiles/hello-upload] Error 1
CMakeFiles/Makefile2:205: recipe for target 'CMakeFiles/hello-upload.dir/all' failed
make[6]: *** [CMakeFiles/hello-upload.dir/all] Error 2
CMakeFiles/Makefile2:212: recipe for target 'CMakeFiles/hello-upload.dir/rule' failed
make[5]: *** [CMakeFiles/hello-upload.dir/rule] Error 2
Makefile:170: recipe for target 'hello-upload' failed
make[4]: *** [hello-upload] Error 2
CMakeFiles/arduino-ros_firmware_hello-upload.dir/build.make:57: recipe for target 'CMakeFiles/arduino-ros_firmware_hello-upload' failed
make[3]: *** [CMakeFiles/arduino-ros_firmware_hello-upload] Error 2
CMakeFiles/Makefile2:520: recipe for target 'CMakeFiles/arduino-ros_firmware_hello-upload.dir/all' failed
make[2]: *** [CMakeFiles/arduino-ros_firmware_hello-upload.dir/all] Error 2
CMakeFiles/Makefile2:527: recipe for target 'CMakeFiles/arduino-ros_firmware_hello-upload.dir/rule' failed
make[1]: *** [CMakeFiles/arduino-ros_firmware_hello-upload.dir/rule] Error 2
Makefile:357: recipe for target 'arduino-ros_firmware_hello-upload' failed
make: *** [arduino-ros_firmware_hello-upload] Error 2
```  

Any help or pointers would be appreciated--thanks!

trov4, 2021-01-12:
Update:
After digging WAY to deep for nothing, I found out that you need to set `SERIAL` in the cmake function `generate_arduino_firmware`, so the full function for me is: 
```cmake 
generate_arduino_firmware(hello
  SRCS chatter.cpp ${ROS_LIB_DIR}/time.cpp
  BOARD uno
  PORT /dev/ttyS4
  SERIAL picocom /dev/ttyS4
)
```  
It uploads correctly for me and runs by adding that line. 

Would it be possible to update the tutorial? Not sure if this is the right place for that.
-T