ros-drivers/rosserial #466: Can't make the package : undefined reference to PyImport:_ImportModule

qori-aziz, 2020-02-19:
Hi, i am sorry that i am new to ROS and its package. When i try to make this package, it says:

`[ 92%]Linking CXX executable /catkin_build_ws/devel/lib/rosserial_server/socket_node`
`/catkin_build_ws/devel/lib/librosserial_server_lookup.so: undefined reference to 'PyImport_ImportModule`
`/catkin_build_ws/devel/lib/librosserial_server_lookup.so: undefined reference to PyString_FromString`
`/catkin_build_ws/devel/lib/librosserial_server_lookup.so: undefined reference to Py_Finalize`
`/catkin_build_ws/devel/lib/librosserial_server_lookup.so: undefined reference to Py_Initialize`
`/catkin_build_ws/devel/lib/librosserial_server_lookup.so: undefined reference to PyString_AsString`
`/catkin_build_ws/devel/lib/librosserial_server_lookup.so: undefined reference to PyObject_GetAttrString`

As i understand, this might be caused by linking fail. Am i right? and how to solve this issue? Thanks in advance

mikepurvis, 2020-02-19:
Yeah, as of #446, `rosserial_server` now links to Python to handle dynamically loading message definitions. This should be set up in the CMakeLists here:

https://github.com/ros-drivers/rosserial/blob/e5195ccb311877f450ea6630422e6d54eadad99c/rosserial_server/CMakeLists.txt#L34-L36

Can you describe more about the environment which is having this issue? What OS, what Python version, what build tool are you using, and how are you invoking it, etc?

qori-aziz, 2020-02-19:
I use ubuntu mate 18 with ROS melodic, in my Raspi 3B+. I build it using catkin_make -j1. I have two python version, both are 2.7 and 3.6. However, i set my ROS_PYTHON_VERSION as 3 because i need to build some package with it.

Before this error, the catkin_make also fails in finding Python.h but i solved it with including Python lib directories in rosserialserver/CMakeLists.txt 
`include_directories(
    include
    ${Boost_INCLUDE_DIRS}
    ${catkin_INCLUDE_DIRS}
    $(PYTHON2_INCLUDE_DIRS)`

Does this have anything to do with this issue?

qori-aziz, 2020-02-19:
Sorry forgot to mention, another think i did to find Python.h from /rosserial_server/src/msg_lookup.cpp was changing `#include "Python.h"` to `#include "python3.6/Python.h"`

mikepurvis, 2020-02-19:
If you're changing `ROS_PYTHON_VERSION` from the default, you have to build your whole workspace from scratch, including all dependencies up to `catkin` itself. You may also need to set `PYTHON_EXECUTABLE=/usr/bin/python3`. Trying to mix and match pythons within a workspace requires a fair bit of care.

mikepurvis, 2020-04-17:
Please feel free to reopen this ticket if the issue persists.

closed 2020-04-17 by mikepurvis