ros-drivers/rosserial #487: Policy CMP0038

rbonghi, 2020-04-15:
Hi all,

I am testing the rosserial_arduino, but I have this warning when I load an Arduino library and I build my code with catkin_make.
A simple example is compile [Measuring Temperature](http://wiki.ros.org/rosserial_arduino/Tutorials/Measuring%20Temperature)

There is something that I lost to add in my CMakeList.txt or a warning that I cannot fix?

```
CMake Warning (dev) at /opt/ros/melodic/share/rosserial_arduino/arduino-cmake/cmake/Platform/Arduino.cmake:986 (add_library):
  Policy CMP0038 is not set: Targets may not link directly to themselves.
  Run "cmake --help-policy CMP0038" for policy details.  Use the cmake_policy
  command to set the policy and suppress this warning.

  Target "uno_Wire" links to itself.
Call Stack (most recent call first):
  /opt/ros/melodic/share/rosserial_arduino/arduino-cmake/cmake/Platform/Arduino.cmake:1043 (setup_arduino_library)
  /opt/ros/melodic/share/rosserial_arduino/arduino-cmake/cmake/Platform/Arduino.cmake:510 (setup_arduino_libraries)
  CMakeLists.txt:32 (generate_arduino_firmware)
This warning is for project developers.  Use -Wno-dev to suppress it.
```

The cmake that I used is:
```cmake
cmake_minimum_required(VERSION 2.8.3)

include_directories(${ROS_LIB_DIR})

# Remove this if using an Arduino without native USB (eg, other than Leonardo)
add_definitions(-DUSB_CON)

generate_arduino_firmware(temperature
  SRCS temperature.cpp ${ROS_LIB_DIR}/time.cpp
  BOARD uno
  PORT /dev/ttyACM0
)
```
Thank you in advance

mikepurvis, 2020-04-17:
Can you confirm that the problem still exists if you patch 5d5191954c63dbb5e58bbae5335586990a0043d4 onto `melodic-devel`?

rbonghi, 2020-04-24:
Hi @mikepurvis ,

Yes, I tried and I think to have a similar message, at this time without policy message warning:
```
CMake Error at /home/raffaello/rosserial_ws/src/rosserial/rosserial_arduino/arduino-cmake/cmake/Platform/Arduino.cmake:986 (add_library):
  Target "uno_Wire" links to itself.
Call Stack (most recent call first):
  /home/raffaello/rosserial_ws/src/rosserial/rosserial_arduino/arduino-cmake/cmake/Platform/Arduino.cmake:1043 (setup_arduino_library)
  /home/raffaello/rosserial_ws/src/rosserial/rosserial_arduino/arduino-cmake/cmake/Platform/Arduino.cmake:510 (setup_arduino_libraries)
  CMakeLists.txt:8 (generate_arduino_firmware)
```
In this case only the target it is linked twice.

I hope I helped you. If there are other tests to do keep me posted .