ros-drivers/rosserial #504: rosserial_python serial_node: No module named 'SerialClient'

shmulike, 2020-07-09:
Hello
I am using Ubuntu Focal 20.04, with Ros Noetic (default python3)
I downloaded the rosserial source package: 'noetic-devel'

I uploaded the example for hello worlds to Arduino Nano.

When I try to run:
`rosrun rosserial_python serial_node.py /dev/ttyUSB0`

```
I receive that message:
Traceback (most recent call last):
  File "/home/me-ubuntu/catkin_ws/devel/lib/rosserial_python/serial_node.py", line 15, in <module>
    exec(compile(fh.read(), python_script, 'exec'), context)
  File "/home/me-ubuntu/catkin_ws/src/rosserial/rosserial_python/nodes/serial_node.py", line 39, in <module>
    from rosserial_python import SerialClient, RosSerialServer
  File "/home/me-ubuntu/catkin_ws/devel/lib/python3/dist-packages/rosserial_python/__init__.py", line 34, in <module>
    exec(__fh.read())
  File "<string>", line 1, in <module>
ModuleNotFoundError: No module named 'SerialClient'
```

Please advise what to do?

mikepurvis, 2020-07-10:
The port of rosserial_python to py3 is still in PR limbo on #480. Myself or someone will needs to cherry-pick the relevant commits forward to the current noetic branch.

shmulike, 2020-07-12:
Do you know maybe when the package will be updated?

My only options left is to downgrade to Ubuntu18 and Melodic

ruffner, 2020-07-22:
I have found that putting a . in front of SerialClient in  ```rosserial_python/src/rosserial_python/__init__.py``` so that it reads:
```from .SerialClient import *``` 
fixes the SerialClient import error

shmulike, 2020-07-22:
Yes, I did that, this is one of the changes from Puthon2 to Ptuhon3

Now I am facing another problem that I don't know how to solve. I am getting an error of unrecognized type: **_basestringI_.**
There are saw several questions about that, some of them suggested replacing it with **_byte_** or something like that.
None of them was working for me.

I know that ROS is open-source, but rosserial is a very basic package.
I really hope that the package will be updated soon.
Thanks to all of you who are making it happen !!

ruffner, 2020-07-23:
To fix that problem, you have to turn *basestring* to just *str*. Then the line after that needs an addition of ```encode()```  added to the end of data  so it reads 
```
elif isinstance(data, str):
    self._write(data.encode())
```

However,  even if you fix this part the code runs, as I have been working on for the past two days, there are still some intricacies with the new stringIO in python3 (I think, maybe the way the byte strings are formed too?) that prohibit the code from actually working. It will run and connect to the serial port and bytes are added to the write queue but for some reason the bytes that are sent are not getting out of the serial port right.

 I thought I was gonna be cool and setup noetic on a raspberry pi but I banged my head against getting rosserial to work for two days straight and all it took was dropping back to melodic and it worked flawlessly off the bat. I too await the noetic support release of rosserial

tkazik, 2021-02-19:
Is this solved? I am also getting the same error on Noetic.

smaakash, 2021-07-24:
In order to fix the serial module not found, you will need to apt install python3-serial. Did you build noetic from source?