ros-drivers/rosserial #512: rosserial server not published on noetic

vincentrou, 2020-09-02:
The rosserial server 0.9.0 does not build on the build farm for noetic.

http://repositories.ros.org/status_page/ros_noetic_default.html?q=serial
http://build.ros.org/view/Nbin_uF64/job/Nbin_uF64__rosserial_server__ubuntu_focal_amd64__binary/1/console

It seems to be related to the pthread lib that is not found.
It build find for me on Ubuntu 20.04.

mikepurvis, 2020-09-02:
Indeed, here's the failure:

```
18:10:17 /usr/bin/ld: CMakeFiles/cmTC_b6678.dir/src.c.o: in function `main':
18:10:17 ./obj-x86_64-linux-gnu/CMakeFiles/CMakeTmp/./obj-x86_64-linux-gnu/CMakeFiles/CMakeTmp/src.c:11: undefined reference to `pthread_create'
18:10:17 /usr/bin/ld: ./obj-x86_64-linux-gnu/CMakeFiles/CMakeTmp/./obj-x86_64-linux-gnu/CMakeFiles/CMakeTmp/src.c:12: undefined reference to `pthread_detach'
18:10:17 /usr/bin/ld: ./obj-x86_64-linux-gnu/CMakeFiles/CMakeTmp/./obj-x86_64-linux-gnu/CMakeFiles/CMakeTmp/src.c:13: undefined reference to `pthread_join'
18:10:17 collect2: error: ld returned 1 exit status
```

I suspect with the breakup of the boost rosdep, we now need to explicitly build/exec depend on `libboost-thread-dev` and `libboost-thread`.

closed 2020-09-06 by mikepurvis

vincentrou, 2020-09-21:
It does not build better with the fix.

Could it be related to the cmake version bump ?

https://stackoverflow.com/questions/1620918/cmake-and-libpthread

Tobias-Fischer, 2020-10-13:
Could this be re-opened? rosserial-server is still not published on noetic

PR #513 (2020-09-06): Add boost::thread dependency to rosserial_server
Should fix #512.