ros-drivers/rosserial #515: Maintainers wanted

mikepurvis, 2020-09-09:
Hey all, I think the community's interest in using and contributing code to rosserial has for a while now outstripped my ability to stay on top of timely review, merges, and releases.

Of those who have been contributing over the last little bit, are there any interested in taking some of this on? You'd get write access to the repo, but obviously your own changes or changes from your company/lab/org should still go through other reviewers.

Pinging especially @stertingen, @asukiaaa, @PetteriAimonen, @rikba, but open to all volunteers.

VedantParanjape, 2020-10-07:
Hi mike, 
I would like to contribute.

asukiaaa, 2020-10-31:
Thank you for offering right of maintaining.
I have a suggestion about library for Arduino.

It may become complicated so I created other issue thread.
Could you consider this?
https://github.com/ros-drivers/rosserial/issues/531