ros-drivers/rosserial #518: cstring: No such file or directory

shmulike, 2020-09-11:
Hello
I'm using Arduino IDE 1.8.13 and Arduino Uno
When I try to verify/compile the HelloWorld example I'm getting this error in the IDE:

```
In file included from /home/user/Arduino/libraries/ros_lib/std_msgs/Time.h:7:0,
                 from /home/user/Arduino/libraries/ros_lib/ros/node_handle.h:40,
                 from /home/user/Arduino/libraries/ros_lib/ros.h:38,
                 from /home/user/Arduino/libraries/ros_lib/examples/HelloWorld/HelloWorld.pde:6:
/home/user/Arduino/libraries/ros_lib/ros/msg.h:40:10: fatal error: cstring: No such file or directory
 #include <cstring>
          ^~~~~~~~~
compilation terminated.
exit status 1
Error compiling for board Arduino Uno.
```

jagyer, 2020-09-13:
I have exactly the same problem! Exactly!
Arduino IDE 1.8.14, Ubuntu 20.04 LTS, ROS Noetic.
Installing rosserial-arduino using binaries just began to work a few days ago (as described in 2.1.1): 
http://wiki.ros.org/rosserial_arduino/Tutorials/Arduino%20IDE%20Setup
Seems like it at least, no errors. And now this, it's so frustrating...
Any advice much appreciated!
Would it help to post the verbose compilation output message from Arduino IDE? It contains a lot...

shmulike, 2020-09-13:
Hi
I am familiar with how to install the ros_lib for Arduino libraries. I was doing it since ROS Indigo.

dmitriypoyarkov, 2020-09-22:
Me too, waiting if there will be somebody to solve this.
In file included from /home/fc/Arduino/libraries/ros_lib/std_msgs/Time.h:7:0,
                 from /home/fc/Arduino/libraries/ros_lib/ros/node_handle.h:40,
                 from /home/fc/Arduino/libraries/ros_lib/ros.h:38,
                 from /home/fc/catkin_ws/src/common_canonical/led_strip_control/firmware/ledstrip_ros/ledstrip_ros.ino:2:
/home/fc/Arduino/libraries/ros_lib/ros/msg.h:40:10: fatal error: cstring: No such file or directory
 #include <cstring>
          ^~~~~~~~~
compilation terminated.

RuslanNig, 2020-09-22:
the same problem

fadamsyah, 2020-09-26:
Hi
I have the same issue

Guasch5000, 2020-09-28:
Maybe this can help a bit.

Remove folder ros_lib inside libraries directory.
Then got to Arduino IDE->Sketch->Include Library->Manage Libraries... ( Ctrl+Shift+I )
And search for "rosserial" and install: Rosserial Arduino Library by Michael Ferguson

Using version 0.7.9, worked with fresh new Arduino Uno and Arduino Mega. Also able to compile for Seeeduino Zero (integrated on Odyssey X86J4105)

fadamsyah, 2020-10-01:
The #525 works well for me finally .... It's very helpful !

jagyer, 2020-10-07:
Both the above tips worked for me, when using the Arduino IDE.
Modifying the msg.h as described helped, and also gave a working ros_lib in Arduino IDE.
However, when using CMake with rosserial_arduino as described here:
http://wiki.ros.org/rosserial_arduino/Tutorials/CMake
it fails when you run catkin_make.
So I'm still stuck with Melodic!
:-(

xVanTuring, 2021-01-14:
@jagyer 
Please Try this , using patch file to update `msg.h` automatically
simple add this line 
```
execute_process(COMMAND patch ${CMAKE_CURRENT_BINARY_DIR}/ros_lib/ros/msg.h ${CMAKE_CURRENT_SOURCE_DIR}/patch/msg.patch)
```
 after `rosserial_configure_client` command in your CMakeLists.txt
and create file `msg.patch` under your project_folder/patch with content:
``` patch
40c40
< #include <cstring>
---
> #include <string.h>
68c68
<     std::memcpy(&val, &f, sizeof(val));
---
>     memcpy(&val, &f, sizeof(val));
182c182
<     std::memcpy(f, &val, sizeof(val));
---
>     memcpy(f, &val, sizeof(val));

```

At least for now...

RBinsonB, 2021-01-21:
> The #525 works well for me finally .... It's very helpful !

I can confirm as well, this was the solution working for me.

jagyer, 2021-02-05:
Still no success with rosserial_arduino using Cmake for me :-/ I can build Arduino code under ROS Noetic using the Arduino IDE and ros_lib from Rosserial Arduino Library by Michael Ferguson version 0.7.9. However, since I'm also using Raspberry Pi running Ubuntu Server 20.04 and ROS Noetic now (works great!), in headless mode, it would be really convenient to be able to build arduino firmware using the Cmake tool in ROS from commandline!  
I just tried out the proposed solution by xVanTuring above. It makes a lot of sense, and it was so clearly described that even I should be able to understand, but ... nope.
I will try it out out soon on a completely "untouched" system also. I might have screwed things up on my system by trying various solution suggestions. But "everything" else in ROS, except for rosserial_arduino, seems to function. Maybe I'm just missing something "obvious"...?
It's a pity though that developers haven't fixed this yet, in the official updates! As it seems...Last installation of Noetic I made was just a couple of weeks ago. Same issue.... But I still have my laptop with Melodic on it, and it works as it should!

viso951, 2021-03-13:
> @jagyer
> Please Try this , using patch file to update `msg.h` automatically
> simple add this line
> 
> ```
> execute_process(COMMAND patch ${CMAKE_CURRENT_BINARY_DIR}/ros_lib/ros/msg.h ${CMAKE_CURRENT_SOURCE_DIR}/patch/msg.patch)
> ```
> 
> after `rosserial_configure_client` command in your CMakeLists.txt
> and create file `msg.patch` under your project_folder/patch with content:
> 
> ```diff
> 40c40
> < #include <cstring>
> ---
> > #include <string.h>
> 68c68
> <     std::memcpy(&val, &f, sizeof(val));
> ---
> >     memcpy(&val, &f, sizeof(val));
> 182c182
> <     std::memcpy(f, &val, sizeof(val));
> ---
> >     memcpy(f, &val, sizeof(val));
> ```
> 
> At least for now...

This worked for me, thank you very much!

WilliBobadilla, 2021-04-04:
> Maybe this can help a bit.
>
> Remove folder ros_lib inside libraries directory.
> Then got to Arduino IDE->Sketch->Include Library->Manage Libraries... ( Ctrl+Shift+I )
> And search for "rosserial" and [...]
this worked for me! thanks a lot

OttawaSTEM, 2021-07-25:
Try this.
In libraries\Rosserial_Arduino_Library\src\ros\msg.h
replace #include<cstring> with the follow code and change romove std:: of  std::memcpy
`
#include <string.h>
void memcpy(void *dest, void *src, size_t n)
{
   char *csrc = (char *)src;
   char *cdest = (char *)dest;
   // Copy contents of src[] to dest[]
   for (int i=0; i<n; i++)
       cdest[i] = csrc[i];
}
`

adarsha4, 2022-04-01:
> Hello I'm using Arduino IDE 1.8.13 and Arduino Uno When I try to verify/compile the HelloWorld example I'm getting this error in the IDE:
>
> ```
> In file included from [...]

Instead of using newer version of rosserial, i used older version 0.7.9. It worked for me

mertbacaksiz, 2023-01-29:
> @jagyer Please Try this , using patch file to update `msg.h` automatically simple add this line
>
> ```
> execute_process(COMMAND patch ${CMAKE_CURRENT_BINARY_DIR}/ros_lib/ros/msg.h [...]

it works thank you

Aman-rkl, 2024-02-07:
It really works , Thanks .