ros-drivers/rosserial #519: Can not export msg and srv in rosserial_arduino by make_libraries.

tongtybj, 2020-09-23:
When I tried `rosrun rosserial_arduino make_libraries.py ~/` with the latest version of `noetic-devel`,
 I got following unexpected result about `rosserial_arduino` package:
```
Exporting rosserial_arduino

  Messages:
Unable to build message: rosserial_arduino/Adc
    Adc,Exporting rosserial_arduino

  Services:
Unable to build service: rosserial_arduino/Test
    Test,

Exporting rosserial_mbed

  Messages:
    Adc,

  Services:
    Test,

Exporting rosserial_msgs

  Messages:
    TopicInfo,Log,

  Services:
    RequestParam,
```
Then, I traced the commit history, and found that before commit  a7ac6cd5f55401148a8e8c854d9029f3e815d8e5, `rosserial_arduino/Adc.msg` and `rosserial_arduino/Test.srv` can be correctly generated in ros_lib. 
Then I checked which changed file  induce such problem. Surprisingly, when I reverted the change of `rosserial_python/src/rosserial_python/SerialClient.py ` in this commit, the problem disappeared. 

Can anyone else (@mikeodr  @asukiaaa ) try following commands to confirm whether this is a common phenomenon or my personal problem?

`git checkout noetic-devel # latest version`
`rosrun rosserial_arduino make_libraries.py ~/  # please check there is no rosserial_arduino inside ros_lib`

`git checkout a7ac6cd`
`rosrun rosserial_arduino make_libraries.py ~/  # please check there is no rosserial_arduino inside ros_lib`

`git reset HEAD~`
`git checkout rosserial_python/src/rosserial_python/SerialClient.py`
`rosrun rosserial_arduino make_libraries.py ~/  # please check there is  rosserial_arduino inside ros_lib`

mikeodr, 2020-09-23:
I think you mean @mikepurvis and not myself. 

I am no longer active with ROS presently. Best of luck with your issue.

tongtybj, 2020-09-24:
@mikeodr 

Thank you for your reply and sorry to bother you.

tongtybj, 2020-09-24:
@mikepurvis 

I narrow down this problem, and found following related error:

```
root@fb70210db6ac:~/ros/rosserial_ws# python3 -c 'import rosserial_arduino'
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/root/ros/rosserial_ws/devel/lib/python3/dist-packages/rosserial_arduino/__init__.py", line 34, in <module>
    exec(__fh.read())
  File "<string>", line 1, in <module>
ModuleNotFoundError: No module named 'SerialClient'
```
This is related to #504, and is fixed by #521