ros-drivers/rosserial #526: Use rosserial with Seeeduino Zero

juliamp22, 2020-10-15:
Hi all,
Does anyone tried to use this package with Seeeduino Zero board?
I have tried to use it without success. The communication between board an rosserial cannot be established

Best Regards,

Júlia

ceinem, 2022-08-06:
I ran into this problem as well, did you find a solution?

juliamp22, 2022-08-07:
Hello Ceinem, 
This problem has two possible ways to be solved:

Option A. If you haven't uploaded the serial library of seedstudio and you haven't bricked the board. Then install the new version of the library.

Option B. If you have bricked the board. You have to buy an  AVR programmer with JTAG, weld wires to the board and upload the bootloader original. https://forum.seeedstudio.com/t/odyssey-x86j4105-ubuntu-20-4-no-dev-ttyacm0-and-no-seeeduino-listed-under-lsusb/254322/10

If you have any doubt, do not hesitate to contact me.

Best,

Júlia

ceinem, 2022-08-07:
Dear Júlia, 
Thanks for your quick reply! 
Could you specify a bit more what you mean by the seeedstudio serial library? a rosserial library by seeedstudio or an actual serial library? Sorry, it's my first time using the Seeeduino. 

It didn't seem to be bricked, as I could without a problem upload new sketches.
However, after attempting this for an hour, I gave up and tossed it back into the drawer and used an Arduino Nano instead for the project.