ros-drivers/rosserial #528: Actions in Arduino code

adbidwai, 2020-10-23:
Is there any way I can write an Action in an Arduino code?

mikepurvis, 2020-10-23:
Under the hood, [an action is just a group of five topics](http://wiki.ros.org/actionlib/DetailedDescription#Action_Interface_.26_Transport_Layer), so although there's no "native" rosserial action server, it wouldn't be terribly difficult to write one yourself.

_However_, I would issue a strong caution on this— ROS and especially rosserial have no mechanism for guaranteeing receipt, and it's extremely easy for an action server/client pair to get out of sync. I have seen this happen a lot in the real world with ROS 1 nodes on separate machines separated by wi-fi; it typically manifests as a deadlock where the server won't accept new goals because it is waiting on a cancellation that the client thinks has already been sent. As nuisancesome as it is, long term you'll probably have a better time if you define your own lossy-tolerant comms scheme between the MCU and computer, and then run a small C++ or Python node on the computer side to interact with those messages and expose the action server.

adbidwai, 2020-10-23:
@mikepurvis Are Service servers/clients are avoided to be used in Arduino for the same reason ?

mikepurvis, 2020-10-26:
Generally yes; this is why they aren't implemented in `rosserial_server`. However, if you are aware of the limitations and have your own timeouts enabled, it's probably possible to do it safely.

Abhinandan-Krishnan, 2021-05-19:
@adbidwai Hey, any workaround that you figured out that can be shared? I too need an Action server on Arduino