ros-drivers/rosserial #537: Cstring breaks old workflows

takacsjd, 2021-01-04:
I'm using older pre 2020 workflows and tutorials for rosserial on arduino @ "Effective programing with ROS"

In msg.h, Arduino does not have a cstring.h. I attempted to rectify this to my environment, but there also other changes in 2020 that utilize this. 

Looking though Google, users are bypassing by using older versions or importing a Cplusplus standard library into arduino. I think recent changes may necessitate a update to the wiki on what libraries need installed - more dependancys are necessary than are pulled when made.

PeterQFR, 2021-01-24:
I had to downgrade rosserial_arduino_library to 0.7.9. I did not find instructions on using a newer version or other CPP standard libraries.

takacsjd, 2021-01-24:
> I had to downgrade rosserial_arduino_library to 0.7.9. I did not find instructions on using a newer version or other CPP standard libraries.
> In order to make this work. I think that updating the instructions to install the required libraries. Can you provide those instructions in a comment here?

I actually just resolved this tonight doing the same after realizing on the ros official wiki for this driver,  they call out the jade branch for use with kinetic (which I'm using on a Raspi, ubiquity robotics free image).

I downloaded that branch, Catkin_make and install out of my workspace then moved Roslib to my arduino libraries and used rosserial make liberaies there. Downloaded to the arduino and had hello world working fine after this.

I take it that (and hope) if you have the latest ros distribution and use the latest rosserial branch, that these items get included when you use the catkin make and install. 

What ROS distro are you using Peter? I'd like to work this issue though

takacsjd, 2021-01-24:
Looks like commit: "Use std::memcpy instead of void casts. (#509)" in 2020 started the cstring dependency. I think my next steps are seeing if cstring is sourced in notic, the latest ROS branch. If it is, then I can close. If not then the dependency chain needs placed into rosserial for when it is made.

PeterQFR, 2021-01-24:
No I'm using Noetic 20.04.

It think the issue could be related to my use of Aduino Due hardware previously. I think in certain circumstances, the library manager gets confused with Libraries compiled against the c std lib for the SAM boards(of which Due is one). I also had c std lib issues with an adafruit library as well. 
After cleaning those libraries out, I was able to correct a cstdlib issue I had with Adafruit, but still have an issue with 0.9.1 of the rosserial Arduino library

Samu24, 2021-03-02:
This is the same issue as #518. There is also a proposal to fix this in #525