ros-drivers/rosserial #541: Release rosserial-server in noetic

Tobias-Fischer, 2021-01-22:
While most packages are released in noetic, rosserial-server is still outstanding. It would be great if this could be done - I think https://github.com/ros-drivers/rosserial/commit/adb84cc931064ea8fc48a081345c48d5d73dc647 fixed the build issue so it should be straightforward.

mikepurvis, 2021-01-22:
Unfortunately, that change was part of 0.9.1 and it still failed to build. I haven't got the bandwidth to dig further at this time, but I'd be delighted if you or someone else could.

Tobias-Fischer, 2021-01-22:
I see. I can have a look if you could point me to the failing build log?

mikepurvis, 2021-01-22:
It's likely been cycled out at this point, but you should be able to reproduce it locally in a container by running a prerelease test: http://wiki.ros.org/bloom/Tutorials/PrereleaseTest

viso951, 2021-03-12:
I am using rosserial 0.9.1 and I have a question, Do you think is possible to launch a rosnode via wifi with an arduino uno wifi rev2?

PR #544 (2021-04-01): Add python3-dev as build depend
Fix #541

Without this fix PythonLibs (`find_package(PythonLibs REQUIRED)`) is not found. Fix verified locally using a prerelease_job.
closed 2021-04-01 by mikepurvis