ros-drivers/rosserial #560: python2 to python3 switch in ros serial

nixinator, 2021-04-25:
I'm having problems with melodic , it's seems to run things here using python2, i could not get roslaunch/rosrun to execute these scripts using python3. This might be environmental problems on the jetson. However python3/python2 are available on the system.

git checkout 8.0 , solves the problem , as these scripts are still python2 compatible. :-)

Would it be possible to have better descriptions in change log of 8.0 to say that was the last release that supported python2?

It's quite possible the i'm just another victim of 

[image]

warnerjon12, 2022-03-04:
I think most of the scripts use /usr/bin/env python (or /.../bin/python). So you can use python3 by making /usr/bin/python a link to python3 rather than python2.

nixinator, 2022-03-04:
I'll give that a try... thanks for tips.

symbolic links, i used them 30 years ago, i'm not sure if i'm happy or sad that they are still in use today.