ros-drivers/rosserial #561: rosrun rosserial_python serial_node.py not work

Keddie-yang, 2021-04-28:
ROS version: Melodic
Operating System: Ubuntu 18.04 Bionic
Processor: Xavier AGX

Question: 
When I use 
```
rosrun rosserial_python serial_node.py _port:=/dev/ttyACM0 _baud:=9600
```  
error messages show:
```
Traceback (most recent call last):
  File "/home/agx/vehicle_ws/src/rosserial/rosserial_python/nodes/serial_node.py", line 39, in <module>
    from rosserial_python import SerialClient, RosSerialServer
  File "/home/agx/vehicle_ws/devel/lib/python2.7/dist-packages/rosserial_python/__init__.py", line 34, in <module>
    exec(__fh.read())
  File "<string>", line 1, in <module>
  File "/home/agx/vehicle_ws/src/rosserial/rosserial_python/src/rosserial_python/SerialClient.py", line 41, in <module>
    import queue
ImportError: No module named queue
```

seyit05, 2021-12-02:
i have same problem could you help?

grammers, 2021-12-10:
I hade the same issue. I changed `import queue` to `from Queue import Queue` in the file `/home/agx/catkin_ws/src/rosserial/rosserial_python/src/rosserial_python/SerialClient.py`