ros-drivers/rosserial #574: ros_lib_molidic (build fail )  in platform io

niboon39, 2021-11-29:
[image]

romainreignier, 2021-11-30:
I do not use mbed anymore but from what I understand from [this document](https://github.com/ARMmbed/mbed-os/blob/f609a5234e91cd450021cd52cf0f6aa881fbfedc/drivers/docs/serial/serial.md), the `BufferedSerial` class is now part of the official mbed API, defined in [drivers/include/drivers/BufferedSerial.h](https://github.com/ARMmbed/mbed-os/blob/f609a5234e91cd450021cd52cf0f6aa881fbfedc/drivers/include/drivers/BufferedSerial.h).
The `rosserial_mbed` should be refactor to use these new interfaces and remove the dependency of the third-party library.

niboon39, 2021-11-30:
Thank you . I can use it

closed 2021-11-30 by niboon39

romainreignier, 2021-12-02:
Hi @niboon39 
I am glad that you could fix your issue.
Could you provide some explanations on how you have managed to fix it. This might help others.
Maybe a change should be made to the `rosserial_mbed` package?

niboon39, 2021-12-03:
When you genarate project in platformio. Add platform_packages = framework-mbed @ ~6.51504.0 in "platformio.ini" 

Tutorial_ROS_MBED
# https://github.com/niboon39/Tutorial_ROS_MBED.git