ros-drivers/rosserial #575: Teensy 4.0 usb serial broken on 20.04?

aatb-ch, 2022-01-06:
I have tried setting up a Teensy 4.0 over usb serial and not getting any luck going through the USB serial connection. It works when going through the hardware Serial1 via a usbserial dongle. Seems to be a 20.04 usb issue, works on 18.04.

Wondered if anybody else has hit this problem? I've tried all the recommended steps, 00-rules, disabling modemmanager, no chance. Works completely fine with a Teensy LC on 20.04, same code, cables. With the T4.0 the serial connection hangs and must physically disconnect it.

warnerjon12, 2022-03-04:
I have used the Teensy 4.0 successfully with rosserial recently on 20.04. Don't think this is still an issue? Make sure you have your udev rules configured correctly.

aatb-ch, 2022-03-04:
After much troubleshooting here https://forum.pjrc.com/threads/69166-Teensy-4-0-usb-serial-broken-on-Ubuntu-20-04?mode=hybrid the issue seems to be stemming from running 20.04 inside a VM environment (Parallels), so likely a usb emulation problem. I resorted to running on a non-VM machine and it all works fine.

closed 2022-03-04 by aatb-ch