ros-drivers/rosserial #597: error: 'void ros::ServiceClient<MReq, MRes>::call(const MReq&, MRes&) [with MReq = rosserial_arduino::TestRequest; MRes = rosserial_arduino::TestResponse]' marked 'override', but does not override    virtual void call(const MReq & request, MRes & response) override

anferjo, 2022-07-29:
Hi guys I'm having some  problems with the ServiceClient example from the Arduino roslib library, once I try to compile it I get the next error message:
  
In file included from /home/user/Arduino/libraries/ros_lib/ros/node_handle.h:62:0,
                 from /home/user/Arduino/libraries/ros_lib/ros.h:38,
                 from /home/user/Arduino/libraries/ros_lib/examples/ServiceClient/ServiceClient.pde:5 :

/home/user/Arduino/libraries/ros_lib/ros/service_client.h: In instantiation of 'class ros::ServiceClient<rosserial_arduino::TestRequest, rosserial_arduino::TestResponse>':
/home/user/Arduino/libraries/ros_lib/examples/ServiceClient/ServiceClient.pde:12:57:   required from here
/home/user/Arduino/libraries/ros_lib/ros/service_client.h:57:16: error: 'void ros::ServiceClient<MReq, MRes>::call(const MReq&, MRes&) [with MReq = rosserial_arduino::TestRequest; MRes = rosserial_arduino::TestResponse]' marked 'override', but does not override
   virtual void call(const MReq & request, MRes & response) override
                ^~~~
exit status 1
Error compiling for board Arduino Mega or Mega 2560.

I'm not sure if this is a problem related to Arduino or with roslib. I generate the lib as it says in the rosserial Arduino tutorial ide setup
( http://wiki.ros.org/rosserial_arduino/Tutorials/Arduino%20IDE%20Setup ) with the next command:

rosrun rosserial_arduino make_libraries.py .

what I'm trying to do is create a client with my own service but I can't even run the client example. Thanks in advance for your cooperation.

sozgundemir, 2023-01-10:
I am having the same issue. Were you able to find any solution?

octo-vaibhav, 2023-01-27:
I have also faced the same issue. After surfing through internet, i came to know that, client in rosserial_arduino has different process to build. following link will help you to create client for arduino.
link: [http://wiki.ros.org/rosserial_client/Tutorials](url)
(PS; I have not tried the tutorial given in the link.) 
I will update the result, if i happen to try the tutorial.

Aistark-web, 2024-01-15:
This is a source code writing error. Because the base class does not have the **call** function, using override is incorrect
Modify the content of the header file **service_client. h** and compile successfully.
Before modification:
`virtual void call(const MReq & request, MRes & response) override`
Remove the **override** keyword.