ros-drivers/rosserial #616: [ERROR] [1524089506.982801]: Unable to sync with device; possible link problem or link software version mismatch such as hydro rosserial_python with groovy Arduino

FlynnDowey, 2023-03-20:
Trying to debug this error that I am getting from https://wiki.ros.org/rosserial_arduino/Tutorials/Hello%20World. I have attempted several suggestions from the web but none have worked. I am using an Arduino Leonardo and the version of ros is noetic. 

Note the following solutions have not worked:

- changing the baud rate
- modifying SerialClient.py and node_handle.h 
- reinstalling rosserial and ros_lib

The error message comes after uploading the HelloWorld example to the Arduino and executing rosrun rosserial_python serial_node.py /dev/ttyACM0.

chamemilo, 2023-05-09:
Did you manage to find the solution?

JinK2000, 2023-05-09:
me too!  STM32F407+ubuntu20.04(notetic)

- Only in rare cases does it work：

[INFO] [1683649672.768868]: Requesting topics...
[INFO] [1683649673.098597]: Note: publish buffer size is 512 bytes

- Most cases report an error like this and i don't know how to solve:

[INFO] [1683650864.691406]: Requesting topics...
[ERROR] [1683650879.695435]: Unable to sync with device; possible link problem or link software version mismatch such as hydro rosserial_python with groovy Arduino

StohanzlMart, 2023-06-09:
Anyone ever got a solution to this? I too have this problem.

I installed ros-noetic-rosserial and ros-noetic-rosserial-arduino and also tried the hello world sketch.

StohanzlMart, 2023-06-10:
Just solved it. Got nothing to do with version mismatch. ros.h has some funky defaults (see https://github.com/ros-drivers/rosserial/issues/354#issuecomment-360833873)

My solution: modify ros.h in lib/ros_lib/ for my hardware to use correct header and thus not try tcp but serial.... (for my full rant see: https://github.com/ros-drivers/rosserial/issues/398#issuecomment-1585569021)

StohanzlMart, 2023-06-10:
Btw, I'm using PlatformIO (an extension to a real IDE (VSC) not Arduino notepad app) and followed this instruction: https://sites.duke.edu/memscapstone/using-rosserial-to-setup-a-ros-node-on-a-teensy/