ros-drivers/rosserial #630: Does the author have any plans to support ROS2？

tianzehao, 2024-03-05:
As described in the title.

gavanderhoorn, 2025-08-14:
This appears to be a duplicate of #589.

ramonvp, 2026-10-01:
For anyone interested, I made a ROS2 port for my own use [here](https://github.com/ramonvp/ros2serial). I wouldn't say it's complete yet, but it works fine with a simple Arduino Uno.

I'd like to finish implementing some missing basic functionality, but I'm having trouble finding enough free time these days. So, if anyone is interested in giving me a hand, I'd greatly appreciate it!

mikepurvis, 2026-10-02:
I'm not the original author of this, but as a former maintainer, no. Part of the pitch with rosserial was maintaining the exact native serialization of the ROS 1 messages and just bridging them over to 8 bit microcontrollers verbatim. I think this is less valuable in 2026 for a variety of reasons:

- rosserial was always awkward for how you integrate it with ros_control, where the expected hardware interface is not ROS topics but rather a C++ plugin that writes into shared data structures on one side and on the other that speaks whatever register-based protocol the hardware natively uses, typically something like canopen, ethercat, dynamixel, etc. The direct ROS message publishing was most interesting and useful for stuff like Log integration, dynamic reconfigure, or transmitting native sensor_msgs such as Range and IMU.
- Another pain point for rosserial was always MD5 mismatch, and ROS 2 still hasn't really delivered a solution there. For something crossing a boundary between such different platforms, you really want to be able to version your protocol and build it on something that lets both sides adapt to the other being slightly ahead or behind rather than that scenario becoming a protocol mismatch error and dead air.
- Again related to ros-control, rosserial never tried to be safe or realtime, so it was immediately a nonstarter for any system with those kinds of requirements.
- And finally, a lot of hobby boards now have 32 bit MCUs and direct network connectivity. If not actual DDS implementations, these devices at least have other options for participating in ROS networks, like websockets passing around JSON blobs.

Sorry to be so negative; it may well be that there is still a place for something rosserial-shaped in ROS 2, but these are gaps from my pov as someone who worked with rosserial and shipped multiple products that heavily used it (especially the Clearpath Jackal and Ridgeback platforms).